analogdevicesinc/iio-oscilloscope #5: runRemote

vladsolov, 2015-01-28:
Thnaks for the previous fast responses! Very impressive support!!!
I am sorry if the next question is the absent of my understanding ; 
It is said on the wiki page that the osc app can be run remotely, is it mean i can run linaro with osc on my host and get the data from the card on the target? what must be installed on the target to make it pass data to the host?
Thanks a lot...

Running Remote
=============
The application can be used to connect to another platform that has a connected device in order to configure the device and  !!!! read data from it  !!!! .

pcercuei, 2015-01-28:
Settings -> Connect
Then just enter the IP address of the target and click OK.

vladsolov, 2015-01-28:
ok, but what do i need to run on the target? must i run linaro desktop or maybe linaro nano or may i use any other os? must i install osc with gui and x11 on the target?

pcercuei, 2015-01-28:
You need https://github.com/analogdevicesinc/libiio installed on the target (both the library and the iiod server).
We only support our linaro image, although it should work as well with other distributions.

vladsolov, 2015-02-01:
Thanks for the response; i got the library and the iiod installed on the targer;
Could u pls advice on how to make the iiod pass data to osc on other machine?
thanks;

pcercuei, 2015-02-02:
On the target, you need a kernel with IIO support.
Please don't use bug reports to ask for help; use https://ez.analog.com instead.

closed 2015-02-02 by pcercuei