analogdevicesinc/iio-oscilloscope #6: Add -Wextra to makefile...

rgetz, 2015-02-01:
If I add 

--- a/Makefile
+++ b/Makefile
@@ -30,7 +30,7 @@ LDFLAGS := $(shell $(PKG_CONFIG) --libs $(DEPENDENCIES)) \
 
 CFLAGS := $(shell $(PKG_CONFIG) --cflags $(DEPENDENCIES)) \
        -I$(SYSROOT)/usr/include $(if $(WITH_MINGW),,-fPIC) \
-       -Wall -g -std=gnu90 -D_GNU_SOURCE -O2 -DPREFIX='"$(PREFIX)"' \
+       -Wall -Wextra -g -std=gnu90 -D_GNU_SOURCE -O2 -DPREFIX='"$(PREFIX)"' \
        -DFRU_FILES=\"$(FRU_FILES)\" -DOSC_VERSION=\"$(GIT_BRANCH)-g$(GIT_HASH)\"
 
I get lots of new warnings. Most are:
plugins/debug.c: In function ‘spinbtn_input_cb’:
plugins/debug.c:609:79: warning: unused parameter ‘data’ [-Wunused-parameter]

in various files, since we don't always use the all the functions necessary by the GTK callbacks.

It's easy to go through and do things like:

+++ b/oscmain.c
@@ -14,12 +14,14 @@ extern GtkWidget *tooltips_en;
 extern GtkWidget *main_window;
 extern struct iio_context *ctx;
 
-static void infobar_hide_cb(GtkButton *btn, gpointer user_data)
+static void infobar_hide_cb(GtkButton __attribute__ ((unused)) *btn,
+               gpointer __attribute__ ((unused)) user_data)
 {
        gtk_widget_set_visible(infobar, false);
 }
 
and quiet things down again, to get the real issues (like this):

plugins/debug.c: In function ‘reg_read_clicked’:
plugins/debug.c:416:2: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]

But the real question is -- does anyone want me to? 

Thanks
-Robin

lclausen-adi, 2015-02-02:
We could probably enable a subset of -Wextra, but e.g. the unused function parameter is way to noisy.

pcercuei, 2015-02-02:
The Makefile in the master branch has an "analyze" target that will use the static analyzer built in LLVM/Clang to discover a plethora of bugs (potential division by zero, memory leaks, mallocs of size zero, uninitialized variables, etc).

rgetz, 2015-04-02:
I will add a few from -Wextra, and close this.

-Robin

closed 2015-04-02 by rgetz