analogdevicesinc/iio-oscilloscope #7: allow 2 FFTs to be run, and displayed at the same time, and be able to measure phase difference on markers

rgetz, 2015-04-02:
I was playing with the branch, and added something like this:

index 34e9372..4af5fc1 100644
--- a/osc.h
+++ b/osc.h
@@ -49,6 +49,7 @@ struct marker_type {
        int bin;
        bool active;
        char label[6];
+       float angle;
        GtkDataboxGraph *graph;
 };
 
diff --git a/oscplot.c b/oscplot.c
index b3fcf64..b7f00f8 100644
--- a/oscplot.c
+++ b/oscplot.c
@@ -984,6 +984,13 @@ static void do_fft(Transform *tr)
                                markers[j].y = (gfloat)out_data[markers[j].bin];
 
                        }
+                       if (fft->num_active_channels == 2) {
+                               markers[j].angle = atan2(settings->imag_source[markers[j].bin],
+                                               in_data[markers[j].bin]);
+
+                               //printf("%i: %f\n", j, markers[j].angle);
+                       } else
+                                markers[j].angle = 0;
                }
                if (*settings->markers_copy) {
                        memcpy(*settings->markers_copy, settings->markers,

This seems to work (angle is set properly). However -- I'm not sure:
 - that markers are turned on for both ffts?
 - where I can get access to both marker data sets? the way transforms are done - is all a little puzzeling to me now.

Thanks
-Robin

rgetz, 2015-04-02:
While I'm thinking about it. The angle on a single FFT/marker can not be averaged. 

The difference between 2 angles can (and should) be averaged.

- Robin

dNechita, 2015-04-02:
Only one set of markers exists and it's owned by the plot. The transforms are independent objects and can only hold the reference of the set.
This means that while capturing, only one transform can fill the markers with data. It was designed that way because we used to have only one FFT/Cross-Correlation/Constellation per plot at a time.
I'm working on changing this so that every transform has its own set of markers, then the plot will have to select which set of markers to display.

Transforms are created when the capture button is pressed and destroyed when the capture stops.
The number of transform created and their type depends on which channels are enabled and the plot domain.
The list of transforms is hold by the "transform_list" variable declared inside "struct _OscPlotPrivate". Through this you can access the transforms and their markers.

-Dan

rgetz, 2015-04-02:
Ok - I guess the important thing is that we make sure that the angles are calculated from the same capture buffer. (When things are in sync in the time domain).

I will leave the above to you. 

Any idea how long this might take?

dNechita, 2015-04-02:
I will finish things no later than today.
I'm going to use the patch you posted to get the angle and then show the difference between angles (the average actually).

rgetz, 2015-04-02:
Sounds great. Looking forward to it. 

The way I was thinking is that the active/front FFT would be the only one controlling marker types (fixed, peak, images, etc), and the 2nd FFT would just use the same bin locations as the first. 

The only issue is how to control which is "first" or "active". There doesn't seem to be any controls of which color is which voltage channels, or which is drawn first. 

- Robin

dNechita, 2015-04-02:
Having one active FFT to control the marker types was something that I also had in mind.
But does it matter which is active since we're interested only in the difference between 2 FFTs?
Do we want to switch the FFT on which the markers apply? (while capturing is on/off)?

rgetz, 2015-04-02:
It should not matter, as long as we always know which is which. 

Otherwise lead/lag will be wrong.

dNechita, 2015-04-02:
I've pushed my changes to the branch:
https://github.com/analogdevicesinc/iio-oscilloscope/tree/phase_difference_2_fft_method

The angle difference is averaged by using a low-pass filter (same way FFT magnitude is displayed when avg > 0). If I should use the conventional average please let me know.

rgetz, 2015-04-03:
It looks great -- I pushed some tweaks to the same branch.

The only thing to double check is if the args to atan2f(y,  x); are in the correct order. We seem to be measuring the correct angles (tested on FMCOMMS5) but I didn't look in too much detail if the sign was right  -90 should lag, and +90 should lead.

See:
http://www.amplifier.cd/Tutorial/Phasenverschiebung/images/slew005.jpg
phase of (channel2, compared to channel 1) is -45 (since channel 2 is lagging channel 1)

If we can double check that - I would just commit move to HEAD, and close.

Thanks
-Robin

rgetz, 2015-04-03:
I spoke too soon.

This seems to work only if you have a time domain plot up, with a trigger running (at least on FMCOMMS5 -- I will test on something else tomorrow). I'm not sure how those are related?

-Robin

dNechita, 2015-04-03:
Those things should not be related.
Are the angle values wrong? or they don't show up at all?

rgetz, 2015-04-03:
The numbers show up, but they look almost random. 

If something has a 90degree offset - it can be +/- 3 degrees, but what I see is almost random.

dNechita, 2015-04-03:
When trigger is enabled and a threshold, rising/falling edge are set the captured signal always has the same phase. I've displayed one of the angles instead of the difference and with the trigger on, the angle value is almost constant. With the trigger off  the angle value appears random and it should. But the difference between angles should not be random.

So are we not measuring the difference right? Should we directly measure the angle difference without replying on time samples?
phase (in radians) = atan(fft_y1(18)/fft_y2(18))?

rgetz, 2015-04-03:
Dan:

I'm not sure it has anything to do with that -- the samples that the fft capture are still random, and it doesn't matter if the time domain plot is on/off. 

It could be an MCS sync issue specific to FMCOMMS5 -- I will look at it a little more. and let you know what I find. I remember talking to Michael about something where he said he does automatic multi-chip sync, but I don't think I was pushing the button to force it. (which could be the problem) -- it was getting late last night.

I will poke around the weekend, and let you know. I think it is pretty close.

-Robin

rgetz, 2015-04-05:
When I make sure that MCS is asserted -- I don't think I see the issue, so I think this is just an FMCOMMS5 issue.

I need to tweak the math, as a little more playing in MATLAB, lead me to re-write the description, so we can provide an more accurate number all the time. (so that we have to do fewer iterations/corrections). I fixed the docs, and now just need to implement & commit things tonight.

-Robin

rgetz, 2015-04-06:
I pushed the code changes - but on ZC706/FMCOMMS5 - was periodically getting double frees and other stack errors/crashes.

Do you see this on this branch?

-Robin

dNechita, 2015-04-06:
I can't see any of your commits on the "phase_difference_2_fft_method" branch.
I gave it a try on the ZC706/FMComms5 anyway but there were no crashes.

-Dan

rgetz, 2015-04-07:
sorry - I had screwed up my local repo, and was committing to a detached state, and it wasn't being pushed upstream. I fixed it, and pushed things up. It seems to work pretty well for me (most of the time).

However - the more I use it, the more I think the MCS doesn't have anything to do with it. It just may be the way libiio works. I will need to check with Paul (or maybe he can comment here). If I grab voltage [01] and [45] - will things make two captures, or one? it seems to make 2, when I'm not playing things back in the time domain...?? does that make sense?

-Robin

rgetz, 2015-04-07:
I just noticed this was branched from the 2014R2 branch, not master. 

Did we want to merge to master or the branch?

dNechita, 2015-04-08:
Sorry it's my mistake, I should have branched it from master. But anyway this changes will eventually end up to both (master and 2014_R2) branches.

rgetz, 2015-04-11:
Dan:

Again, so I am testing with FMCOMMS5, and channels [0,1] and [4,5]

So, I switched to : 

root@analog:/usr/local/src/iio-oscilloscope# git status
On branch clean_phase_diff_2_fft_method
Your branch is up-to-date with 'origin/clean_phase_diff_2_fft_method'.

and added:

root@analog:/usr/local/src/iio-oscilloscope# git diff
diff --git a/oscplot.c b/oscplot.c
index 779596d..5d68487 100644
--- a/oscplot.c
+++ b/oscplot.c
@@ -2258,6 +2258,7 @@ static void markers_phase_diff_show(OscPlotPrivate *priv)
        struct marker_type *trB_markers;
        float angle_diff, lead_lag;
        float filter, angle;
+       float tmp;
 
        gtk_text_buffer_set_text(priv->phase_buf, "", -1);
        gtk_text_buffer_get_iter_at_line(priv->phase_buf, &iter, 1);
@@ -2288,10 +2289,15 @@ static void markers_phase_diff_show(OscPlotPrivate *priv)
                                        avg[m] = lead_lag;
 
                                /* Cosine law, answers are [0, +pi] */
-                               angle_diff =  acosf((crealf(trA_markers[m].vector) * crealf(trB_markers[m].vector) +
-                                               cimagf(trA_markers[m].vector) * cimagf(trB_markers[m].vector)) /
-                                               (cabsf(trA_markers[m].vector) * cabsf(trB_markers[m].vector))) * 180 / M_PI;
 
+                               tmp =  (crealf(trA_markers[m].vector) * crealf(trB_markers[m].vector) +
+                                               cimagf(trA_markers[m].vector) * cimagf(trB_markers[m].vector)) /
+                                               (cabsf(trA_markers[m].vector) * cabsf(trB_markers[m].vector));
+                               angle_diff = acosf(tmp);
+printf("%2.3f [%2.3f + %2.3f] * [%2.3f + %2.3f] : [%2.3f %2.3f] %2.3f %2.3f : ", lead_lag, 
+               crealf(trA_markers[m].vector), cimagf(trA_markers[m].vector),
+               crealf(trB_markers[m].vector), cimagf(trB_markers[m].vector),
+               cabsf(trA_markers[m].vector), cabsf(trB_markers[m].vector), tmp, angle_diff);
                                /* put back into the correct quadrant */
                                if (lead_lag < 0)
                                        angle_diff *= -1.0;
@@ -2299,6 +2305,7 @@ static void markers_phase_diff_show(OscPlotPrivate *priv)
                                if (lead_lag > 180.0)
                                        angle_diff = 360.0 - angle_diff;
 
+printf("%2.3f\n", angle_diff);
                                avg[m] = ((1 - filter) * avg[m]) + (filter * angle_diff);
 
                                angle = avg[m];

and it prints out problems...

15.616 [-409.000 + 437.000] * [-269.000 + 517.000] : [598.540 582.795] 0.963 0.273 : 0.273
14.410 [-426.000 + -415.000] * [-516.000 + -296.000] : [594.728 594.871] 0.969 0.252 : 0.252
14.526 [447.000 + 391.000] * [532.000 + 267.000] : [593.877 595.242] 0.968 0.254 : 0.254
39.353 [561.000 + -195.000] * [308.000 + -503.000] : [593.924 589.808] 0.773 0.687 : 0.687
67.634 [491.000 + 339.000] * [491.000 + -319.000] : [596.659 585.527] 0.381 1.180 : 1.180
-43.781 [432.000 + -414.000] * [0.000 + 0.000] : [598.348 0.000] nan nan : -nan

you can see, it prints out ~14 some times, but (a) is screws up (prints out 39 & 67 - meaning there is still a problem is looking at the way fft's are captured, and sometimes' the points are [0 + 0*i], meaning something is also messed up.

So - there are still some issues to be fixed.

doing something stupid like:

adding:
root@analog:/usr/local/src/iio-oscilloscope# git diff
diff --git a/oscplot.c b/oscplot.c
index 779596d..c00ea01 100644
--- a/oscplot.c
+++ b/oscplot.c
@@ -765,7 +765,7 @@ static void do_fft(Transform *tr)
        unsigned int maxX[MAX_MARKERS + 1];
        gfloat maxY[MAX_MARKERS + 1];
        gfloat plugin_fft_corr;
-
+printf("%s\n", __func__);
        if (settings->marker_type)
                marker_type = *((enum marker_types *)settings->marker_type);
 

indicates:

do_fft
do_fft
-166.556 [-562.000 + -186.000] * [587.000 + 50.000] : [591.980 589.126] -0.973 2.907 : -2.907
-165.002 [208.000 + -553.000] * [-58.000 + 590.000] : [590.824 592.844] -0.966 2.880 : -2.880
192.907 [-163.000 + 568.000] * [32.000 + -590.000] : [590.926 590.867] -0.975 2.916 : 357.084
193.974 [409.000 + 424.000] * [-495.000 + -310.000] : [589.115 584.059] -0.970 2.898 : 357.102
194.940 [547.000 + 218.000] * [-588.000 + -70.000] : [588.840 592.152] -0.966 2.881 : 357.119
101.706 [-121.000 + 584.000] * [0.000 + 0.000] : [596.403 0.000] nan nan : nan
do_fft
do_fft
194.675 [535.000 + 276.000] * [-572.000 + -128.000] : [601.997 586.147] -0.967 2.885 : 357.115
193.846 [-285.000 + 525.000] * [149.000 + -570.000] : [597.369 589.153] -0.971 2.900 : 357.100
-166.124 [254.000 + -535.000] * [-117.000 + 574.000] : [592.234 585.803] -0.971 2.899 : -2.899
-166.061 [-345.000 + -488.000] * [446.000 + 385.000] : [597.636 589.187] -0.971 2.898 : -2.898
2.294 [196.000 + -560.000] * [173.000 + -566.000] : [593.309 591.849] 0.999 0.040 : 0.040
-149.737 [-509.000 + -297.000] * [0.000 + 0.000] : [589.313 0.000] nan nan : -nan
do_fft
do_fft
-165.441 [-285.000 + -522.000] * [399.000 + 425.000] : [594.734 582.946] -0.968 2.887 : -2.887
-164.973 [530.000 + -275.000] * [-434.000 + 397.000] : [597.097 588.188] -0.966 2.879 : -2.879
194.578 [-514.000 + 301.000] * [417.000 + -416.000] : [595.648 589.020] -0.968 2.887 : 357.113
-39.341 [8.000 + 593.000] * [-366.000 + 459.000] : [593.054 587.058] 0.773 0.687 : -0.687
26.001 [-480.000 + 356.000] * [-271.000 + 522.000] : [597.609 588.154] 0.899 0.454 : 0.454
64.249 [260.000 + 539.000] * [0.000 + 0.000] : [598.432 0.000] nan nan : nan

which is obviously wrong. There should be 2 do_fft, and then it should print out one, and only one line of angles, and then 2 do_ffts again....

so, there are stil some major bugs for you to figure out before I can go forward.

-Robin

mhennerich, 2015-04-13:
Hi Robin,

The patch quoted here is scrambled. 
Things work for me. Only adding to print statements see below.

The extra lines come from the extra markers.
As far as I can tell, the angle_diff always computes proper differences, the lead_lag doesn't seem to be necessary.
 

diff --git a/oscplot.c b/oscplot.c
index 779596d..65cb80f 100644
--- a/oscplot.c
+++ b/oscplot.c
@@ -766,6 +766,8 @@ static void do_fft(Transform *tr)
        gfloat maxY[MAX_MARKERS + 1];
        gfloat plugin_fft_corr;
 
+       printf("%s\n", __func__);
+
        if (settings->marker_type)
                marker_type = *((enum marker_types *)settings->marker_type);
 
@@ -2292,6 +2294,9 @@ static void markers_phase_diff_show(OscPlotPrivate *priv)
                                                cimagf(trA_markers[m].vector) * cimagf(trB_markers[m].vector)) /
                                                (cabsf(trA_markers[m].vector) * cabsf(trB_markers[m].vector))) * 180 / M_PI;
 
+
+                               printf("lead_lag %f angle_diff %f\n",lead_lag,angle_diff);
+
                                /* put back into the correct quadrant */
                                if (lead_lag < 0)
                                        angle_diff *= -1.0;

do_fft
do_fft
lead_lag -74.967430 angle_diff 74.967430
do_fft
do_fft
lead_lag -74.967430 angle_diff 74.967430
do_fft
do_fft
lead_lag 285.033997 angle_diff 74.966011
do_fft
do_fft
lead_lag -74.983154 angle_diff 74.983154
do_fft
do_fft
lead_lag -75.026733 angle_diff 75.026733
do_fft
do_fft
lead_lag -75.038513 angle_diff 75.038521
do_fft
do_fft
lead_lag -74.983154 angle_diff 74.983154
do_fft
do_fft
lead_lag 285.033997 angle_diff 74.966011
do_fft
do_fft
lead_lag -74.983154 angle_diff 74.983154
do_fft
do_fft
lead_lag -75.038803 angle_diff 75.038818
do_fft
do_fft
lead_lag -75.024773 angle_diff 75.024780
do_fft
do_fft
lead_lag -74.958870 angle_diff 74.958870
do_fft
do_fft
lead_lag 285.011871 angle_diff 74.988098

rgetz, 2015-04-13:
Oh, and I only had one marker turned on. 

- Robin

rgetz, 2015-04-13:
The angle issues are described at:
http://wiki.analog.com/resources/eval/user-guides/ad-fmcomms2-ebz/iq_rotation#determining_phase_difference

If the angle is greater than 180 degrees, the cosine angle will always calculate the inner angle, and give you the wrong number. 

You need the value based on tan() to determine if the difference is between 0 and 180 or 180 and 360.

rgetz, 2015-04-14:
I rebooted the board, and for some reason that fixed all the random issues I was seeing. (things were hung in the HDL, and I couldn't capture data anymore by the time I rebooted).

After a little more tweaking of the math (committed), things work pretty well for me.

Michael - can you test, and then Dan can move to master?

-Robin

rgetz, 2015-04-21:
Dan:

Can you merge into master?

Thanks
-Robin

dNechita, 2015-04-21:
Merged: https://github.com/analogdevicesinc/iio-oscilloscope/commit/5c544cb021a1c287d95055472e34cd2179491bc9

closed 2015-04-21 by dNechita