analogdevicesinc/iio-oscilloscope #10: osc crashes when running test.ini files

rgetz, 2015-04-15:
When I add something to an ini file, which looks like:

[IIO Oscilloscope - Capture Window3]
cycle = 200
save_markers = log.txt
quit = 1

it crashes:
Program received signal SIGSEGV, Segmentation fault.
0xb6bebf9c in iio_context_get_device () from /usr/lib/libiio.so.0
(gdb) bt
#0  0xb6bebf9c in iio_context_get_device () from /usr/lib/libiio.so.0
#1  0xb6fc56cc in device_list_treeview_init (plot=0x5e09a8) at oscplot.c:3245
#2  create_plot (plot=0x5e09a8) at oscplot.c:6172
#3  osc_plot_init (plot=0x5e09a8) at oscplot.c:436
#4  0xb6c1ffc0 in g_type_create_instance () from /usr/lib/arm-linux-gnueabihf/libgobject-2.0.so.0
#5  0xb6c0b7f2 in ?? () from /usr/lib/arm-linux-gnueabihf/libgobject-2.0.so.0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

when I add this line to the ini file (at the top): 
test = 1

I get a : 
Unable to write value: 12.000000 dB
Sequential loading of profile aborted.

?

pcercuei, 2015-04-15:
This should fix the issue: https://github.com/analogdevicesinc/linux/commit/9999669c5e49ee20a3a9ac8c7afc0cdb3bee1945

rgetz, 2015-04-15:
Does that mean I just need to update the kernel? (adi_update_boot)?

rgetz, 2015-04-17:
New kernel seems to work -- but it would be nice (but not necessary) to add the workaround in the osc app as well. (until the new kernel is uploaded on the web site).

Closing for now.

closed 2015-04-17 by rgetz