analogdevicesinc/iio-oscilloscope #11: when testing fmcomms5, the test hangs

rgetz, 2015-04-16:
root@analog:/usr/local/src/iio-oscilloscope# ./osc -p ./profiles/FMComms5_test.ini 

(osc:6114): Gtk-WARNING **: Could not load image 'osc128.png': Failed to open file '/usr/local/src/iio-oscilloscope/osc128.png': No such file or directory

(osc:6114): Gtk-WARNING **: Could not load image 'IIOlogo.png': Failed to open file '/usr/local/src/iio-oscilloscope/IIOlogo.png': No such file or directory

(osc:6114): Gtk-WARNING **: Could not load image 'ADIlogo.png': Failed to open file '/usr/local/src/iio-oscilloscope/ADIlogo.png': No such file or directory
Fontconfig warning: ignoring C.UTF-8: not a valid language tag

(osc:6114): Gtk-WARNING **: Could not load image 'osc_capture.png': Failed to open file '/usr/local/src/iio-oscilloscope/osc_capture.png': No such file or directory

(osc:6114): Gtk-WARNING **: Could not load image 'osc_generator.png': Failed to open file '/usr/local/src/iio-oscilloscope/osc_generator.png': No such file or directory
Error creating proxy: The connection is closed (g-io-error-quark, 18)
Error creating proxy: The connection is closed (g-io-error-quark, 18)
Error creating proxy: The connection is closed (g-io-error-quark, 18)
Error creating proxy: The connection is closed (g-io-error-quark, 18)
Found plugin: FMComms6
Found plugin: FMComms5
Found plugin: Motor Control
Found plugin: SCPI
Found plugin: FMComms1
Found plugin: DAQ2
Found plugin: FMComms2/3/4/5 Advanced
Found plugin: Debug
Found plugin: FMComms2/3/4
Found plugin: CN0357
Found plugin: AD5628-1
Found plugin: AD6676
Found plugin: AD7303
Found plugin: Partial Reconfiguration
Found plugin: DMM
Loaded plugin: FMComms5
Loaded plugin: FMComms2/3/4/5 Advanced
Loaded plugin: Debug
Loaded plugin: DMM

(osc:6114): Gtk-WARNING **: Could not load image 'ADIlogo.png': Failed to open file '/usr/local/src/iio-oscilloscope/ADIlogo.png': No such file or directory

(osc:6114): Gdk-CRITICAL **: gdk_window_move_resize_internal: assertion 'GDK_IS_WINDOW (window)' failed

(osc:6114): Gdk-CRITICAL **: _gdk_pixmap_new: assertion '(drawable != NULL) || (depth != -1)' failed
Unhandled tokens in ini file, 
	Section IIO Oscilloscope - Capture Window3
	Attribute : Math.expanded
	Value: 0
Unhandled tokens in ini file, 
	Section IIO Oscilloscope - Capture Window3
	Attribute : Math.active
	Value: 0
Loading profile sequentially from /tmp/osc_6114.ini
Restoring detached state for plugin FMComms2/3/4/5 Advanced (0)
Restoring detached state for plugin DMM (0)
Restoring detached state for plugin Debug (0)
Restoring detached state for plugin FMComms5 (0)

(osc:6114): Gtk-WARNING **: Could not load image 'ADIlogo.png': Failed to open file '/usr/local/src/iio-oscilloscope/ADIlogo.png': No such file or directory

(osc:6114): Gdk-CRITICAL **: gdk_window_move_resize_internal: assertion 'GDK_IS_WINDOW (window)' failed

(osc:6114): Gdk-CRITICAL **: _gdk_pixmap_new: assertion '(drawable != NULL) || (depth != -1)' failed
skipping /sys/devices/soc0/fpga-axi@0/41600000.i2c/i2c-0/i2c-3/3-0054/eeprom (size == 1024)
Closing plugin: FMComms5
Closing plugin: FMComms2/3/4/5 Advanced
Closing plugin: Debug
Closing plugin: DMM
Found plugin: FMComms6
Found plugin: FMComms5
Found plugin: Motor Control
Found plugin: SCPI
Found plugin: FMComms1
Found plugin: DAQ2
Found plugin: FMComms2/3/4/5 Advanced
Found plugin: Debug
Found plugin: FMComms2/3/4
Found plugin: CN0357
Found plugin: AD5628-1
Found plugin: AD6676
Found plugin: AD7303
Found plugin: Partial Reconfiguration
Found plugin: DMM
Loaded plugin: FMComms5
Loaded plugin: FMComms2/3/4/5 Advanced
Loaded plugin: Debug
Loaded plugin: DMM

(osc:6114): Gtk-WARNING **: Could not load image 'ADIlogo.png': Failed to open file '/usr/local/src/iio-oscilloscope/ADIlogo.png': No such file or directory

(osc:6114): Gdk-CRITICAL **: gdk_window_move_resize_internal: assertion 'GDK_IS_WINDOW (window)' failed

(osc:6114): Gdk-CRITICAL **: _gdk_pixmap_new: assertion '(drawable != NULL) || (depth != -1)' failed
Unhandled tokens in ini file, 
	Section IIO Oscilloscope - Capture Window3
	Attribute : Math.expanded
	Value: 0
Unhandled tokens in ini file, 
	Section IIO Oscilloscope - Capture Window3
	Attribute : Math.active
	Value: 0

(osc:6114): Gtk-WARNING **: Could not load image 'ADIlogo.png': Failed to open file '/usr/local/src/iio-oscilloscope/ADIlogo.png': No such file or directory

(osc:6114): Gdk-CRITICAL **: gdk_window_move_resize_internal: assertion 'GDK_IS_WINDOW (window)' failed

(osc:6114): Gdk-CRITICAL **: _gdk_pixmap_new: assertion '(drawable != NULL) || (depth != -1)' failed
(test.ad7291.in_temp0_raw.int = 80 160): value = 138
Test passed.
(test.ad7291.in_voltage0_raw.int = 1578 2087): value = 2079
Test passed.
(test.ad7291.in_voltage1_raw.int = 2893 3826): value = 3568
Test passed.
(test.ad7291.in_voltage2_raw.int = 2893 3826): value = 3575
Test passed.
(test.ad9361-phy.in_voltage0_rssi.int = 100 150): value = 112
Test passed.
(test.ad9361-phy.in_voltage1_rssi.int = 100 150): value = 109
Test passed.
(test.ad9361-phy-B.in_voltage0_rssi.int = 100 150): value = 113
Test passed.
(test.ad9361-phy-B.in_voltage1_rssi.int = 100 150): value = 108
Test passed.
(test.ad9361-phy.in_voltage0_hardwaregain.double = 60.0 80.0): value = 71.000000
Test passed.
(test.ad9361-phy.in_voltage1_hardwaregain.double = 60.0 80.0): value = 71.000000
Test passed.
(test.ad9361-phy-B.in_voltage0_hardwaregain.double = 60.0 80.0): value = 71.000000
Test passed.
(test.ad9361-phy-B.in_voltage1_hardwaregain.double = 60.0 80.0): value = 71.000000
Test passed.
(test.ad9361-phy.in_voltage0_rssi.int = 20 50): value = 40
Test passed.
(test.ad9361-phy.in_voltage1_rssi.int = 20 50): value = 40
Test passed.
(test.ad9361-phy-B.in_voltage0_rssi.int = 20 50): value = 40
Test passed.
(test.ad9361-phy-B.in_voltage1_rssi.int = 20 50): value = 41
Test passed.
(test.ad9361-phy.in_voltage0_hardwaregain.double = 0.0 28.0): value = 18.000000
Test passed.
(test.ad9361-phy.in_voltage1_hardwaregain.double = 0.0 28.0): value = 18.000000
Test passed.
(test.ad9361-phy-B.in_voltage0_hardwaregain.double = 0.0 28.0): value = 18.000000
Test passed.
(test.ad9361-phy-B.in_voltage1_hardwaregain.double = 0.0 28.0): value = 19.000000
Test passed.

(osc:6114): Gtk-WARNING **: Could not load image 'ADIlogo.png': Failed to open file '/usr/local/src/iio-oscilloscope/ADIlogo.png': No such file or directory

(osc:6114): Gdk-CRITICAL **: gdk_window_move_resize_internal: assertion 'GDK_IS_WINDOW (window)' failed

(osc:6114): Gdk-CRITICAL **: _gdk_pixmap_new: assertion '(drawable != NULL) || (depth != -1)' failed

(osc:6114): Gtk-CRITICAL **: IA__gtk_widget_hide: assertion 'GTK_IS_WIDGET (widget)' failed

and then it hangs...

-Robin

pcercuei, 2015-04-17:
Do you have markers saved in "log.txt"?

pcercuei, 2015-04-20:
With this commit: https://github.com/analogdevicesinc/iio-oscilloscope/commit/f4d9cac7ab3a3e3c8adae148097eb631dc3aa5c5

And commenting some DMM tests in the profile (which you did as well, according to the log), all the tests now pass and I get to the "ship it" dialog.

rgetz, 2015-04-20:
Thanks.

I will update the DMM/voltage tests in master, and then close this.

-Robin

rgetz, 2015-04-21:
When a test fails, I get a print out on the console (which is necessary),

(test.ad7291.in_voltage3_raw.int = 3459 4005): value = 4095
*** Test failed! ***

 but I don't get the "test failed window" that use to come up. This is necessary during production test, where things are run without a console window open.

There use to be something that called create_blocking_popup() when a test failed, but it looks like you removed it.

https://github.com/analogdevicesinc/iio-oscilloscope/commit/36fa74d9532184f69f547430172a54bc9afab21c#diff-c74b82bc75d5a01887591f7a663f242eL2039

-Robin

pcercuei, 2015-04-21:
It was necessary at that moment, but I think I can reintroduce it without too much effort.

pcercuei, 2015-04-21:
Fixed in https://github.com/analogdevicesinc/iio-oscilloscope/commit/0bb2158576108d8a220e550abc4bfeabd1c5578c

rgetz, 2015-04-21:
almost. The console and the message don't match.

Console says:
(test.ad7291.in_voltage3_raw.int = 3764 3482): value = 4095

Message says:
Test was: test.ad7291.in_voltage3_raw.int = 0 3764
Value read = 0

rgetz, 2015-04-21:
sometimes this crashes as well....

(test.ad7291.in_temp0_raw.int = 80 160): value = 143
Test passed.
(test.ad7291.in_voltage0_raw.int = 1706 2100): value = 2081
Test passed.
(test.ad7291.in_voltage1_raw.int = 3132 3635): value = 3550
Test passed.
(test.ad7291.in_voltage2_raw.int = 3132 3635): value = 3576
Test passed.
(test.ad7291.in_voltage3_raw.int = 3764 3482): value = 4095
[Thread 0xb1aff2d0 (LWP 19242) exited]
[Thread 0xb2cff2d0 (LWP 19243) exited]

Program received signal SIGSEGV, Segmentation fault.
0xb1cee478 in ?? ()
(gdb) bt
#0  0xb1cee478 in ?? ()
#1  0xb6a3aac8 in ?? () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) info threads
  Id   Target Id         Frame 
  15   Thread 0xb0aff2d0 (LWP 19234) "pool" __libc_do_syscall () at ../ports/sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:43
  3    Thread 0xb3e142d0 (LWP 19219) "gmain" 0xb6b6ed20 in poll () at ../sysdeps/unix/syscall-template.S:81
  2    Thread 0xb48862d0 (LWP 19218) "gdbus" 0xb6b6ed20 in poll () at ../sysdeps/unix/syscall-template.S:81
* 1    Thread 0xb5eae000 (LWP 19215) "osc" 0xb1cee478 in ?? ()

pcercuei, 2015-04-21:
Does https://github.com/analogdevicesinc/iio-oscilloscope/commit/bb523ce08ac58eef9b0c14a36a140c8d9565f2e6 fix your problem with the messages not matching?

About the segmentation fault... I've been chasing it for quite some time, it happens **inside** gtk_main_iteration() in fmcomms2_adv.c and it seems like a race with the calibration thread...

rgetz, 2015-04-21:
Yes, that fixes it.

No, this was happening with the FMCOMMS5 test - so there is no cal thread running.

-Robin

closed 2015-04-21 by rgetz

pcercuei, 2015-04-21:
There is - the fmcomms2_adv.c plugin is used for fmcomms5 as well, and the fmcomms5 test profile performs a calibration.

rgetz, 2015-04-21:
You are correct -- FMCOMMS5 does do a cal - sorry, I forgot about that.

-Robin