analogdevicesinc/iio-oscilloscope #15: osc.c:osc_plugin_default_handle() doesn't handle "debug.xxx" tokens

rgetz, 2015-04-17:
Ini files which have "debug.xxxx" tokens in them (like any of the FMCOMMS2/3/4/5 when running as root, when the advanced tab is running) the default handler doesn't know how to deal with them, and it passes them to the plugin, which assumes that default handler took care of it. (since it does in the other non-sequential situation).

I added something in a few places which prints this out if there is an error, rather than silently eating it - which is what it did before.

Thanks
-Robin

pcercuei, 2015-04-17:
Could you test this and tell me if it fixes the issue? https://gist.github.com/pcercuei/65100986e77fc16845b2

rgetz, 2015-04-17:
Seems to work.

Thanks
-Robin

pcercuei, 2015-04-17:
Thanks.
Fixed in https://github.com/analogdevicesinc/iio-oscilloscope/commit/39fb8d8f3513466e70d2ba50689ed106135b7cee

closed 2015-04-17 by pcercuei