analogdevicesinc/iio-oscilloscope #16: quit = 1 in ini file crashes osc

rgetz, 2015-04-17:
if you add a "quit = 1" in the end of an ini file. you get something like:

Program received signal SIGSEGV, Segmentation fault.
0xb6bebf9c in iio_context_get_device () from /usr/lib/libiio.so.0
(gdb) bt
#0  0xb6bebf9c in iio_context_get_device () from /usr/lib/libiio.so.0
#1  0xb6fc5512 in device_list_treeview_init (plot=0x4b27d0) at oscplot.c:3066
#2  create_plot (plot=0x4b27d0) at oscplot.c:6002
#3  osc_plot_init (plot=0x4b27d0) at oscplot.c:432
#4  0xb6c1ffc0 in g_type_create_instance () from /usr/lib/arm-linux-gnueabihf/libgobject-2.0.so.0
#5  0xb6c0b7f2 in ?? () from /usr/lib/arm-linux-gnueabihf/libgobject-2.0.so.0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

dNechita, 2015-04-21:
Fix in commits https://github.com/analogdevicesinc/iio-oscilloscope/commit/98a5c74b70c631a20c07a7b7e2e986c83968efb0, https://github.com/analogdevicesinc/iio-oscilloscope/commit/449b589e7d12c0cf77749b2eece507f50bf2df5c.

dNechita, 2015-04-21:
Reopening issue since there is more work to be done. The commits above do not fix the issue entirely.

pcercuei, 2015-04-21:
I think it's because the "quit" token triggers a call to application_quit(), which destroys the context but doesn't stop the application.

dNechita, 2015-04-21:
It is true.
I'm trying to figure out which is best: to exit the application when inside the application_quit() thus creating more exit points for the program or keep a single exit point and try to make the application to quit gracefully.

rgetz, 2015-04-22:
quit gracefully seems like the better idea (less long term maintenance).

.

rgetz, 2015-05-09:
This mostly works now, so closing. If it's a problem -- I will re-open

closed 2015-05-09 by rgetz