analogdevicesinc/iio-oscilloscope #17: quiting (control-Q) when processing an ini file crashes

rgetz, 2015-04-20:
setup an ini file with a loop, and cycles = something (so it takes some time to run), 

While the ini is being parsed, control-Q (or FILE -> QUIT), and you get something like:

Program received signal SIGSEGV, Segmentation fault.
0xb24ee478 in ?? ()
(gdb) bt
#0  0xb24ee478 in ?? ()
#1  0xb6a3bac8 in ?? () from /lib/arm-linux-gnueabihf/libglib-2.0.so.0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) info threads
  Id   Target Id         Frame 
  11   Thread 0xb0aff2d0 (LWP 15015) "pool" __libc_do_syscall () at ../ports/sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:43
  3    Thread 0xb3e152d0 (LWP 15007) "gmain" 0xb6b6fd20 in poll () at ../sysdeps/unix/syscall-template.S:81
  2    Thread 0xb48872d0 (LWP 15006) "gdbus" 0xb6b6fd20 in poll () at ../sysdeps/unix/syscall-template.S:81
* 1    Thread 0xb5eaf000 (LWP 15003) "osc" 0xb24ee478 in ?? ()

I assume this is because quit is quiting and destroys some references / that the ini parsing is still using.
-Robin

dNechita, 2015-04-20:
It actually destroys the entire plot that the ini parsing is trying to update/use.

I need to rethink the profile loading architecture in order to avoid this scenario.
It's great that we can manipulate (start/stop capture, do loop iteration, quit, measure stuff, etc.) the application through a .ini file but this feature is not without architecture challenges. I'll think of something.

-Dan

dNechita, 2015-04-20:
Something similar happens on issue #16: The application loads a profile and destroys itself when "quit=1" line is encountered followed by an attempt to continue the execution using destroyed data.

-Dan

rgetz, 2015-04-20:
It's not just "great", it's a requirement to to be able to start/stop captures (FFT, time domain, etc), log results (including markers & IIO attributes), control external equipment (via SCPI), iterate over loops, test/compare results, bring up windows, quit (with and without error codes), etc.

We need to make this more robust to be able to allow others to write more scripts. Right now the overall flakiness in ini parsing is stopping me from using the osc app, because I need to debug too many things. 

-Robin