analogdevicesinc/iio-oscilloscope #18: ini processing silently eats things if they are in the wrong section

rgetz, 2015-04-21:
an ini file which contains something like:

<SEQ> i 1000000 1000000 10000000
[FMComms2/3/4/5 Advanced]
cf-ad9361-dds-core-lpc.out_altvoltage0_TX1_I_F1_frequency = <i>
cf-ad9361-dds-core-lpc.out_altvoltage5_TX2_I_F2_frequency = <i>
cf-ad9361-dds-core-lpc.out_altvoltage4_TX2_I_F1_frequency = <i>
cf-ad9361-dds-core-lpc.out_altvoltage6_TX2_Q_F1_frequency = <i>
cf-ad9361-dds-core-lpc.out_altvoltage3_TX1_Q_F2_frequency = <i>
cf-ad9361-dds-core-lpc.out_altvoltage7_TX2_Q_F2_frequency = <i>
cf-ad9361-dds-core-lpc.out_altvoltage2_TX1_Q_F1_frequency = <i>
cf-ad9361-dds-core-lpc.out_altvoltage1_TX1_I_F2_frequency = <i>

[IIO Oscilloscope - Capture Window3]
cycle = 1000
save_markers = log.txt
</SEQ>

works well.

However, if you forget the first "[FMComms2/3/4/5 Advanced]", the osc app silently eats the cf-... lines in the capture window processing without throwing an error.

The user has no idea what went wrong.

-Robin

pcercuei, 2015-04-21:
Is that a problem? The values of the "frequency" attributes will still be applied, so there is no error to throw out...

rgetz, 2015-04-21:
The frequency attributes will not be applied unless they go into the right section. That is the problem.

-Robin

pcercuei, 2015-04-21:
OK you are right, they will be applied in any **plugin** section but not in the main OSC one or in one of a capture window.
I pushed a commit https://github.com/analogdevicesinc/iio-oscilloscope/commit/4fe3ec4154362dd03a2517cb2b0076a5d0c84118 that should fix it.

pcercuei, 2015-04-21:
("fix it" as in, show error messages)

rgetz, 2015-04-21:
I think it works, but now, it gives me other errors - that Dan needs to fix.

Unhandled attribute in main section: window_x_pos = 0
Unhandled attribute in main section: window_y_pos = 25
Unhandled attribute in main section: test = 1

those are handled in osc:load_profile, so they need to be skipped in your checks.

I think it would be fine to print out something on stderr, "unknown ini attribute", with a line number - since this sort of thing is for people developing their own ini files, rather than doing a hard stop of the ini processing.

-Robin

pcercuei, 2015-04-22:
I fixed it: https://github.com/analogdevicesinc/iio-oscilloscope/commit/4cfd2eba76881cd06d648bc6d8d482ae3cc4c700

pcercuei, 2015-04-22:
And the error messages should now display the line number starting from https://github.com/analogdevicesinc/iio-oscilloscope/commit/98ea4aaad240db80c1ee9329429d8018c4608d5f

rgetz, 2015-05-09:
Seems like it mostly works -- If I have any more issues, I will send a patch.

closed 2015-05-09 by rgetz