analogdevicesinc/iio-oscilloscope #20: would be nice to be able to handle nested (at least 2 or 3) loops in ini files

rgetz, 2015-04-21:
The <SEQ> </SEQ> parsing in libini2.c is pretty rudimentary, it would be nice if we could use nested loops.

For example, to sweep the DDS from -10 to +10 MHz, we sweep from 0 to 10, and then 0 to 10 again with the phase set differently.

<seq> phase 0 180000 180000
set phase
<seq> freq 0 1000000 10000000

</seq>
</seq>

Right now, we can't do this.

Same for sweeping frequency over different amplitudes... 

Thanks
-Robin

dNechita, 2015-04-28:
Enhancement is at commit: https://github.com/analogdevicesinc/iio-oscilloscope/commit/63382c0c8cc14c78f81a880ed912d53703dba2d9

However the inner loop does not know about the parent's SEQ variable.
E.g.
<SEQ> x 1 2 10
some_property=<x>
    <SEQ> y 1 1 3
    some_other_property=<y> + <x>    ;<x> is unknown here and will not be replaced
    </SEQ>
</SEQ>

Is this going to be a burden or it doesn't matter?

-Dan

rgetz, 2015-04-28:
I'm not exactly sure what you mean about the inner loop not knowing about outer loop.

What I would like to do is:
<SEQ> mag 0 -1 -10
<SEQ> freq 1000000 1000000 10000000 
scale = <mag>
freq = <freq>
save_markers = log.txt
</SEQ>
</SEQ>

that will not work?

-Robin

dNechita, 2015-04-28:
Yes it will work.

What I mean is that the scope of "mag" variable is limited to the outer loop only.

-Dan

-Dan

dNechita, 2015-04-28:
Now that I've seen the entire example I have to say that it won't work.
But this will work:
{SEQ} mag 0 -1 -10
scale = {mag}
{SEQ} freq 1000000 1000000 10000000 
freq = {freq}
save_markers = log.txt
{/SEQ}
{/SEQ}

-Dan

rgetz, 2015-04-28:
Is there a warning/error? or does it just silently do nothing?

dNechita, 2015-04-29:
I'm working on a more robust implementation and I'll get back to you soon.

-Dan

rgetz, 2015-04-29:
Yeah, the more tests I run, the more limiting this is. We need to be able to use outer vars in inner loops.

dNechita, 2015-04-29:
Variables from outer loops can now be seen by inner loops due to commit: https://github.com/analogdevicesinc/iio-oscilloscope/commit/2bc1e2bbcdc92cf9ca573ce5775fbf7775e16b98

rgetz, 2015-04-29:
Excellent - I will try it out, and let you know

-Robin

rgetz, 2015-05-01:
works for me.

Thanks

closed 2015-05-01 by rgetz