analogdevicesinc/iio-oscilloscope #21: parsing ini files with non-fixed size incremental loops

rgetz, 2015-04-21:
Today we support the ini parsing with the same syntax that the shell command supports.

NAME
       seq - print a sequence of numbers
SYNOPSIS
       seq [OPTION]... FIRST INCREMENT LAST

DESCRIPTION
       Print numbers from FIRST to LAST, in steps of INCREMENT.

It would be nice (like for the FMCOMMS1 test), were we test at different points, but they are not evenly spaced to handle a list, like the 

for i in {1 4 6 99 123}

can do

Some other keyword other than "SEQ" should be used.

may want to think about this while working on #20 .

-Robin

dNechita, 2015-04-30:
Enhancement is at commit https://github.com/analogdevicesinc/iio-oscilloscope/commit/0a0818ea8c4ba0ae907abafb5a6472eece11bd25.

Usage:
\<FOR\> i in {1 4 6 99 123}
\</FOR\> 

You can also embed one or multiple \<FOR\> loops inside a \<SEQ\> loop or the other way around.
The space-separated values of the \<FOR\> loop can be any string(without spaces), not just integers. Something like this should work:

\<FOR\> message in {message1 message2 messge3}
test.message=\<message\>
\</FOR\>

-Dan

rgetz, 2015-04-30:
This is great -- can we add some docs for this in the wiki?

Thanks

dNechita, 2015-04-30:
Will do.

rgetz, 2015-05-09:
Thanks - seems to be 
http://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope_profile_guidelines

Tim : Can you update the FMCOMMS1 test for this? (to make it more readable?)

-Robin

radhermit, 2015-05-11:
Sure, I can look at the FMCOMMS1 tests.

radhermit, 2015-05-21:
I forgot to note, this has been in the fmcomms1-test branch for a while. I'd like to test it before merging but haven't gotten around to setting up the complete hardware configuration required for the tests yet.

rgetz, 2016-09-23:
Since this is implemented - I will close this.

closed 2016-09-23 by rgetz