analogdevicesinc/iio-oscilloscope #22: FMCOMMS5 doesn't load up FIR filters properly from ini files (or doesn't set the widget)

rgetz, 2015-04-22:
When you load an filter, and turn it on, close the osc application, it should store the filter file, and that fact it is "on", in the ini file (which it seems to do).

When starting things up a 2nd time, the ini file is loaded, the filter isn't loaded (or the widget is just left "off") - I think the latter - but if you can check, and fix it -- it would be great.

-Robin

dNechita, 2015-04-22:
The plugins were not being restored at all because of one of my commits: https://github.com/analogdevicesinc/iio-oscilloscope/commit/1c42d07f769a6aeffa8b4fb04b2d59a181e217a1
Fix is at: https://github.com/analogdevicesinc/iio-oscilloscope/commit/4e7cc0e778c747ca64feac358a0df604b28cce4f

closed 2015-04-22 by dNechita

rgetz, 2015-04-22:
Dan:

Did you test the filters on fmcomms2/3/4/5? Even when I was setting the profile by hand - it wasn't restored properly. 

- Robin

dNechita, 2015-04-22:
If by testing you mean that I should have checked if the GUI has been restored properly then yes. I did not check if the filter has some effect on the received data.

dNechita, 2015-04-22:
Found that it was working for me only because I was using the default profile to check for this bug. It's true that things don't get restored properly when loading a profile specified by user (GUI or command line). 
Fix is at https://github.com/analogdevicesinc/iio-oscilloscope/commit/c8efa5068006eee49e6994bfdb1983c76d3a4e7e