analogdevicesinc/iio-oscilloscope #24: windows version crashes

rkutty, 2015-04-24:
latest running from 
http://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope

running in windbg

attached to target, during the "connect" button is pressed:

0:008> g
ModLoad: 000007fe`fce90000 000007fe`fcee5000   C:\Windows\system32\mswsock.dll
ModLoad: 000007fe`fc5b0000 000007fe`fc5b7000   C:\Windows\System32\wshtcpip.dll
ModLoad: 00000000`63e00000 00000000`63e12000   C:\Program Files\IIO Oscilloscope\lib\osc\AD5628_1.dll
ModLoad: 00000000`6a000000 00000000`6a010000   C:\Program Files\IIO Oscilloscope\lib\osc\ad6676.dll
ModLoad: 00000000`614c0000 00000000`614e3000   C:\Program Files\IIO Oscilloscope\lib\osc\AD7303.dll
ModLoad: 00000000`64c00000 00000000`64c0e000   C:\Program Files\IIO Oscilloscope\lib\osc\cn0357.dll
ModLoad: 00000000`041b0000 00000000`041c4000   C:\Program Files\IIO Oscilloscope\lib\osc\daq2.dll
ModLoad: 00000000`6d1c0000 00000000`6d1d2000   C:\Program Files\IIO Oscilloscope\lib\osc\debug.dll
ModLoad: 00000000`67100000 00000000`67110000   C:\Program Files\IIO Oscilloscope\lib\osc\dmm.dll
ModLoad: 00000000`70780000 00000000`707a0000   C:\Program Files\IIO Oscilloscope\lib\osc\fmcomms1.dll
ModLoad: 00000000`6e980000 00000000`6e99a000   C:\Program Files\IIO Oscilloscope\lib\osc\fmcomms2.dll
ModLoad: 00000000`6edc0000 00000000`6edd4000   C:\Program Files\IIO Oscilloscope\lib\osc\fmcomms2_adv.dll
ModLoad: 00000000`6db40000 00000000`6db5a000   C:\Program Files\IIO Oscilloscope\lib\osc\fmcomms5.dll
ModLoad: 00000000`6e940000 00000000`6e94f000   C:\Program Files\IIO Oscilloscope\lib\osc\fmcomms6.dll
ModLoad: 00000000`65dc0000 00000000`65dd2000   C:\Program Files\IIO Oscilloscope\lib\osc\motor_control.dll
ModLoad: 00000000`635c0000 00000000`638ce000   C:\Program Files\IIO Oscilloscope\lib\osc\pr_config.dll
(133c.1384): Access violation - code c0000005 (first chance)
First chance exceptions are reported before any exception handling.
This exception may be expected and handled.
*** ERROR: Symbol file could not be found.  Defaulted to export symbols for C:\Program Files\IIO Oscilloscope\bin\libosc.dll - 
libosc!ini_next_section+0x7:
00000000`6b3989c7 488b4110        mov     rax,qword ptr [rcx+10h] ds:00000000`00000010=????????????????

0:000> k
Child-SP          RetAddr           Call Site
00000000`0022e1f0 00000000`6b3993ae libosc!ini_next_section+0x7
*** ERROR: Symbol file could not be found.  Defaulted to export symbols for C:\Program Files\IIO Oscilloscope\lib\osc\fmcomms2.dll - 
00000000`0022e230 00000000`6e98454f libosc!update_from_ini+0x6e
00000000`0022e2f0 00000000`6b382f06 fmcomms2!handle_external_request+0x40f
00000000`0022e350 00000000`6b38308b libosc+0x2f06
00000000`0022e7b0 00000000`6b384500 libosc+0x308b
00000000`0022e810 00000000`6b3846b4 libosc!check_inifile+0xf0
00000000`0022ec50 00000000`6b39763b libosc!do_init+0x104
*** ERROR: Symbol file could not be found.  Defaulted to export symbols for C:\Program Files\IIO Oscilloscope\bin\libgobject-2.0-0.dll - 
00000000`0022ece0 00000000`63a4603c libosc!cb_connect+0xeb
00000000`0022ed30 00000000`63a57272 libgobject_2_0_0!g_closure_invoke+0x15c
00000000`0022edd0 00000000`63a5f5d9 libgobject_2_0_0!g_signal_handler_disconnect+0xe92
00000000`0022ef30 00000000`63a5f778 libgobject_2_0_0!g_signal_emit_valist+0xca9
*** ERROR: Symbol file could not be found.  Defaulted to export symbols for C:\Program Files\IIO Oscilloscope\bin\libgtk-win32-2.0-0.dll - 
00000000`0022f0c0 00000000`61987f67 libgobject_2_0_0!g_signal_emit+0x18
00000000`0022f100 00000000`61872f2d libgtk_win32_2_0_0!gtk_widget_activate+0x77
00000000`0022f140 00000000`6187330e libgtk_win32_2_0_0!gtk_menu_shell_activate_item+0xfd
00000000`0022f190 00000000`6185e08b libgtk_win32_2_0_0!gtk_menu_shell_activate_item+0x4de
00000000`0022f220 00000000`63a4603c libgtk_win32_2_0_0!gtk_marshal_VOID__UINT_STRING+0x13b
00000000`0022f280 00000000`63a57164 libgobject_2_0_0!g_closure_invoke+0x15c
00000000`0022f320 00000000`63a5f22c libgobject_2_0_0!g_signal_handler_disconnect+0xd84
00000000`0022f480 00000000`63a5f778 libgobject_2_0_0!g_signal_emit_valist+0x8fc
00000000`0022f620 00000000`619892fd libgobject_2_0_0!g_signal_emit+0x18
00000000`0022f660 00000000`6185c261 libgtk_win32_2_0_0!gtk_widget_translate_coordinates+0x4bd
00000000`0022f6d0 00000000`6185c6db libgtk_win32_2_0_0!gtk_propagate_event+0xc1
*** ERROR: Symbol file could not be found.  Defaulted to export symbols for C:\Program Files\IIO Oscilloscope\bin\libgdk-win32-2.0-0.dll - 
00000000`0022f720 00000000`6c37d132 libgtk_win32_2_0_0!gtk_main_do_event+0x39b
*** ERROR: Symbol file could not be found.  Defaulted to export symbols for C:\Program Files\IIO Oscilloscope\bin\libglib-2.0-0.dll - 
00000000`0022f7b0 00000000`685f1b94 libgdk_win32_2_0_0!gdk_win32_drawable_get_handle+0xed2
00000000`0022f7e0 00000000`685f1d88 libglib_2_0_0!g_main_context_dispatch+0x244
00000000`0022f890 00000000`685f2213 libglib_2_0_0!g_main_context_dispatch+0x438
00000000`0022f910 00000000`6185b640 libglib_2_0_0!g_main_loop_run+0xc3
*** WARNING: Unable to verify timestamp for C:\Program Files\IIO Oscilloscope\bin\osc.exe
*** ERROR: Module load completed but symbols could not be loaded for C:\Program Files\IIO Oscilloscope\bin\osc.exe
00000000`0022f960 00000000`00403fc9 libgtk_win32_2_0_0!gtk_main+0xa0
00000000`0022f9f0 00000000`004013e8 osc+0x3fc9
00000000`0022fe60 00000000`004014eb osc+0x13e8
*** ERROR: Symbol file could not be found.  Defaulted to export symbols for C:\Windows\system32\kernel32.dll - 
00000000`0022ff30 00000000`7776652d osc+0x14eb
00000000`0022ff60 00000000`7789c521 kernel32!BaseThreadInitThunk+0xd
00000000`0022ff90 00000000`00000000 ntdll!RtlUserThreadStart+0x21

rgetz, 2015-04-27:
I get the same thing on my machine - it works the very first time, and then never again...

-Robin

muaddib0308, 2015-04-27:
Hello,

I installed IO Scope from the wiki page and was using the tool for ten minutes or so before it crashed.
Once it crashed, I could not reconnect to the RF PICOZED Module.

I uninstalled the software, re-installed it and made an additional attempt.  The software crashes after entering the IP address and clicking 'ok'.

Cheers
Neil

rgetz, 2015-04-29:
I think the crashing is gone, but it still does not seem to connect properly to a remote device. Any updates?

-Robin

pcercuei, 2015-04-30:
Connecting remotely works just fine here, what kind of trouble do you have?

rgetz, 2015-04-30:
On windows - connection just brings up hourglass, and spins, and eventually stops spinning, and doesn't connect. 

As if we typed in a bad IP number.

pcercuei, 2015-04-30:
Sorry, I can't reproduce. It works just fine here. Double-check that you don't have a conflict of IP addresses, and that you can ping the board (or SSH into it).

rgetz, 2015-04-30:
Yes, we can do that (ping and ssh with putty), osc doesn't work here.

Can you call Tim, and get him to reproduce. I'm not in the lab.

muaddib0308, 2015-04-30:
Will do

muaddib0308, 2015-04-30:
Wasn't downloading from Michael's share.

muaddib0308, 2015-04-30:
Tim helped me download the .exe from Michael's 'share'.  IIO scope works, can see CW tone's in time / frequency domain etc.

rgetz, 2015-05-03:
closing - assuming that the working version will be posted publically soon

closed 2015-05-03 by rgetz