analogdevicesinc/iio-oscilloscope #26: osc application does not "remember" local/network context

rgetz, 2015-04-29:
I can't automatically tell the IIO scope to connect to a remote target by giving it an IP number.

I need to be able to do this to test some headless boards.

-Robin

rgetz, 2015-04-30:
Tim worked on this a bit, and made a remote_ip_addr branch for now. Things should just auto-connect if you add a remote_ip_addr field in the main profile section.

It can probably be merged, but there are probably some latent issues I didn't consider where multiple iio contexts may be created since the init process is a bit convoluted.

I will check it out, and get back to him.

-robin

pcercuei, 2015-04-30:
Actually only one IIO context is created, and then it is "cloned": https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/osc.c#L2052

So if you initialize "ctx" early enough the OSC should work just fine.

radhermit, 2015-05-01:
I also added support on that branch for specifying a device to connect to via "osc -c [IP address]" which basically does what Paul mentioned.

rgetz, 2015-05-09:
Command line args are OK, but it doesn't really help out those people running on Windows...

radhermit, 2015-05-11:
That's why I also had the remote_ip_addr ini option too. I'll rebase the branch and post it for review/merging.

radhermit, 2015-05-26:
Closing since the remote_ip_addr ini support was also merged.

closed 2015-05-26 by radhermit