analogdevicesinc/iio-oscilloscope #27: reconnecting multiple times to remote devices segfaults osc

radhermit, 2015-05-01:
Often it crashes on the second connection for me, but sometimes it takes up to four or five reconnects to segfault.

To reproduce, run osc, go to Settings -> Connect, enter a working device's IP address, and connect to it. Then keep reconnecting to the same device until osc segfaults.

Here's an example gdb backtrace from a crash:

backtrace:
#0  0x00007fffe83ee4a0 in ?? ()
No symbol table info available.
#1  0x00007ffff56282c3 in g_timeout_dispatch () from /usr/lib64/libglib-2.0.so.0
No symbol table info available.
#2  0x00007ffff562781d in g_main_context_dispatch () from /usr/lib64/libglib-2.0.so.0
No symbol table info available.
#3  0x00007ffff5627bf8 in g_main_context_iterate.isra () from /usr/lib64/libglib-2.0.so.0
No symbol table info available.
#4  0x00007ffff5627f22 in g_main_loop_run () from /usr/lib64/libglib-2.0.so.0
No symbol table info available.
#5  0x00007ffff7267087 in gtk_main () from /usr/lib64/libgtk-x11-2.0.so.0
No symbol table info available.
#6  0x0000000000401b02 in main (argc=1, argv=0x7fffffffdc98) at oscmain.c:172
        c = 0
        profile = 0x0

registers:
rax            0x0	0
rbx            0x0	0
rcx            0x0	0
rdx            0x0	0
rsi            0x7fffe83ee4a0	140737089823904
rdi            0x0	0
rbp            0x7fffdc0946c0	0x7fffdc0946c0
rsp            0x7fffffffd998	0x7fffffffd998
r8             0x6	6
r9             0x7ffff59168c0	140737313335488
r10            0x70	112
r11            0xe	14
r12            0x641440	6558784
r13            0x68c560	6866272
r14            0x7fffdc0946c0	140736884983488
r15            0x7ffff56282b0	140737310261936
rip            0x7fffe83ee4a0	0x7fffe83ee4a0
eflags         0x10206	[ PF IF RF ]
cs             0x33	51
ss             0x2b	43
ds             0x0	0
es             0x0	0
fs             0x0	0
gs             0x0	0

current instructions:
=> 0x7fffe83ee4a0:	Cannot access memory at address 0x7fffe83ee4a0

threads backtrace:

Thread 3 (Thread 0x7fffeb790700 (LWP 18298)):
#0  0x00007ffff3d5f32d in poll () from /lib64/libc.so.6
#1  0x00007ffff5627b94 in g_main_context_iterate.isra () from /usr/lib64/libglib-2.0.so.0
#2  0x00007ffff5627cac in g_main_context_iteration () from /usr/lib64/libglib-2.0.so.0
#3  0x00007ffff5627ce9 in glib_worker_main () from /usr/lib64/libglib-2.0.so.0
#4  0x00007ffff564e4f5 in g_thread_proxy () from /usr/lib64/libglib-2.0.so.0
#5  0x00007ffff4022334 in start_thread () from /lib64/libpthread.so.0
#6  0x00007ffff3d681ed in clone () from /lib64/libc.so.6

Thread 2 (Thread 0x7fffd27fc700 (LWP 18357)):
#0  0x00007ffff3d63de9 in syscall () from /lib64/libc.so.6
#1  0x00007ffff566c177 in g_cond_wait_until () from /usr/lib64/libglib-2.0.so.0
#2  0x00007ffff55fcb49 in g_async_queue_pop_intern_unlocked () from /usr/lib64/libglib-2.0.so.0
#3  0x00007ffff55fd16b in g_async_queue_timeout_pop () from /usr/lib64/libglib-2.0.so.0
#4  0x00007ffff564ef4c in g_thread_pool_thread_proxy () from /usr/lib64/libglib-2.0.so.0
#5  0x00007ffff564e4f5 in g_thread_proxy () from /usr/lib64/libglib-2.0.so.0
#6  0x00007ffff4022334 in start_thread () from /lib64/libpthread.so.0
#7  0x00007ffff3d681ed in clone () from /lib64/libc.so.6

Thread 1 (Thread 0x7ffff7fc48c0 (LWP 18294)):
#0  0x00007fffe83ee4a0 in ?? ()
#1  0x00007ffff56282c3 in g_timeout_dispatch () from /usr/lib64/libglib-2.0.so.0
#2  0x00007ffff562781d in g_main_context_dispatch () from /usr/lib64/libglib-2.0.so.0
#3  0x00007ffff5627bf8 in g_main_context_iterate.isra () from /usr/lib64/libglib-2.0.so.0
#4  0x00007ffff5627f22 in g_main_loop_run () from /usr/lib64/libglib-2.0.so.0
#5  0x00007ffff7267087 in gtk_main () from /usr/lib64/libgtk-x11-2.0.so.0
#6  0x0000000000401b02 in main (argc=1, argv=0x7fffffffdc98) at oscmain.c:172

pcercuei, 2015-05-04:
What plugins are loaded?

pcercuei, 2015-05-04:
It should be fixed now, could you confirm?

radhermit, 2015-05-04:
Yep, it appears that fixed it, but I've only tested against the plugins loaded for an fmcomms3 board. Thanks!

closed 2015-05-04 by radhermit