analogdevicesinc/iio-oscilloscope #29: Reading information from DMM tab whilst capturing data.

muaddib0308, 2015-05-06:
When I tried to read the temperature of the AD9361 (using the DMM tab) with my plot still capturing data, the tool crashed.

rgetz, 2015-05-06:
This was on the windows version or the Linux version? I think I have done this on the Linux version without issue.

muaddib0308, 2015-05-06:
Windows platform, thanks.

pcercuei, 2015-05-06:
Are you using the alpha release released here: https://github.com/analogdevicesinc/iio-oscilloscope/releases/tag/v0.1-alpha1 ?

muaddib0308, 2015-05-06:
Tim downloaded onto my machine last week, from Michael's share.  

(Analog Devices Inc. Industrial I/O Oscilloscope master-gb4a8aa3 helpful?)

Can download and re-assess

muaddib0308, 2015-05-06:
Did not mean to close. 

Tim downloaded onto my machine, not sure from where... Michael's share?

(Analog Devices Inc. Industrial I/O Oscilloscope master-gb4a8aa3 helpful?)

Will download from your link and re-assess if needed.

pcercuei, 2015-05-06:
I fixed some stability issues in the v0.1-alpha1 one. Nothing related to DMM though, so could you try to see if it indeed only happens when the DMM plugin refreshes?

muaddib0308, 2015-05-06:
Downloaded from the link above and re-checked.  I do not see the error described any longer.

pcercuei, 2015-05-07:
Ok, I'll close the issue then, please re-open if it happens again.

closed 2015-05-07 by pcercuei