analogdevicesinc/iio-oscilloscope #39: An attempt to find peaks of a cross-correlation throws a segmentaion fault error

dNechita, 2015-06-04:
I would like to apply the below patch to the function that finds peaks of a cross-correlation plot but I want to check with you first so that I don't break the peak finder algorithm.
The problem is that i variable starts from 1 and on the first iteration the out_data[i - 2] will be out-of-bounds.
 
``` 
diff --git a/oscplot.c b/oscplot.c
index ac7fcb9..1169da5 100644
--- a/oscplot.c
+++ b/oscplot.c
@@ -1274,19 +1274,19 @@ void cross_correlation_transform_function(Transform *tr, gboolean init_transform
 				maxY[0] = out_data[0];
 			} else {
 				for (j = 0; j <= MAX_MARKERS && markers[j].active; j++) {
-					if  ((fabs(out_data[i - 1]) > maxY[j]) &&
-						((!((out_data[i - 2] > out_data[i - 1]) &&
-						 (out_data[i - 1] > out_data[i]))) &&
-						 (!((out_data[i - 2] < out_data[i - 1]) &&
-						 (out_data[i - 1] < out_data[i]))))) {
+					if  ((fabs(out_data[i]) > maxY[j]) &&
+						((!((out_data[i - 1] > out_data[i]) &&
+						 (out_data[i] > out_data[i + 1]))) &&
+						 (!((out_data[i - 1] < out_data[i]) &&
+						 (out_data[i] < out_data[i + 1]))))) {
 						if (marker_type == MARKER_PEAK) {
 							for (k = MAX_MARKERS; k > j; k--) {
 								maxY[k] = maxY[k - 1];
 								maxX[k] = maxX[k - 1];
 							}
 						}
-						maxY[j] = fabs(out_data[i - 1]);
-						maxX[j] = i - 1;
+						maxY[j] = fabs(out_data[i]);
+						maxX[j] = i;
 						break;
 					}
 				}
```

-Dan

rgetz, 2015-06-04:
That will break things, and cause false positives.

I think you are better off just doing something like:

https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/oscplot.c#L1272

- if (i == 0) {
+ if (i <= 2) {

Same issue exists in the fft:

https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/oscplot.c#905

dNechita, 2015-06-04:
Thanks Robin!

-Dan

closed 2015-06-04 by dNechita