analogdevicesinc/iio-oscilloscope #44: parrallel builds?

rgetz, 2015-07-06:
On the Zynq target:

time make
real	2m46.945s
user	2m37.810s
sys	0m6.040s

time make -j 2
real	1m34.303s
user	2m46.180s
sys	0m6.910s

can we automatically build things with number of CPUs?

radhermit, 2015-07-07:
The easiest way to blanket cover all makefile builds on the target would be to set something like:

export MAKEFLAGS="-j$(nproc)"

in /etc/profile or similar for the image.

rgetz, 2015-07-17:
That's a great ideao, but do we expect people to do that on non-target as well?

time make
real	0m7.384s
user	0m6.204s
sys	0m0.556s

time make -j2
real	0m4.055s
user	0m6.752s
sys	0m0.496s

?

pcercuei, 2015-07-17:
7s vs. 4s, do we really care here?

radhermit, 2015-07-21:
Not really. Especially since the end developer/user compiling stuff is somewhat likely to have more than 2 cores these days.

If we want to, the easiest thing would be to mention something in the osc build docs on the wiki or put some default make flags in the example commands to run.

rgetz, 2016-09-23:
I updated the docs, and am closing this.
https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope

closed 2016-09-23 by rgetz