analogdevicesinc/iio-oscilloscope #53: General (maybe silly) questions

oubejja, 2016-05-13:
Hi everyone!
I hope i'm in the right section.
I began reading various source files since a couple of days, in order to comprehend how data is processed (Rx path) and I have some questions :
- Is the FFTW (and other operations) applied to captured data for processing purposes or plotting purposes only ? or both ?
- If I want to re-use/collect processed data in real-time (spectrum peaks for example) to another software/application running on the host PC (which I'm willing to develop), is it possible ? If so, from which point (or file) should I start reading/modifying. 
If you have any advice, comment or if I didn't make my point clear. I'm all ears =)
Thank you in advance.

rgetz, 2016-05-13:
oubejja:

most of the Q/A happens for us on Engineerzone - 

https://ez.analog.com/community/linux-device-drivers/linux-software-drivers

but we can answer some questions here - there are just less people monitoring the github issue trackers.

Depending on what you want to do - saving markers to a text file is already supported.

Have a look at:
https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/profiles/FMComms4_sweep.ini

at the bottom of the file, there is (really) basic looping, and saving marker data.

This is implemented here:
https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/oscplot.c#L5149

There are also ways to pass markers, or raw data to the other plug ins - depending on what you want to do.

Or - the other option is just to use libiio, and make your application stand alone.

-Robin

oubejja, 2016-05-13:
rgetz :
Thx for the quick answer! I'll try to explain my project :

I want to turn the system : _**ZC706 + FMCOMMS4 + My app** on windows or linux host_ into a "light" spectrum analyzer, that can hopefully scan the desired bands (WiFi 2.4 & 5 GHz, DECT, 3G/4G not simultaneously) by choosing a frequency band window (button click, no tuner) in order to minimize calculations and sweep/plot fast enough. 

IIO Scope does that partially, but it doesn't identify the received frames, as WiFi SSID for instance,  which can be done on laptop with the usual integrated WiFi adapter.

I thought recently about adding a WiFi or 3G adapter directly to the ZC706, I don't know if it's feasible. Maybe you can tell me about this (I hope!).

So anyway in order to advance, I considered collecting data (time and frequency domain) from IIO Scope sources/memory at some **_processing step_**. Or maybe rebuild IIO Scope on Linux host with the modified source code (libiio).

I hope it's clearer now =)
Thx again

-Othmane

rgetz, 2016-09-23:
I think people are doing this now - so closing this

closed 2016-09-23 by rgetz