analogdevicesinc/iio-oscilloscope #54: test failures don't return error codes

rgetz, 2016-07-28:
It use to be when a profile/test was run, and the test failed, the OSC application returned a non-success error code, so that shell scripts would understand what was going on.

This is no longer the case, and causes issues in the test infrastructure.

I don't know who wants to have a look at things (or who has time?)

Thanks
-Robin

radhermit, 2016-07-28:
I've had a patch sitting around for doing vibration tests fully on the command line running osc under Xvfb that probably handles this fix among other things (https://github.com/analogdevicesinc/board-tests/blob/master/picozed-sdr1-brk/vibration/osc-cli-test.patch).

In particular, the return codes from the profile loading part of the patch should fix this. I'll clean it up and push the relevant changes.

rgetz, 2016-07-29:
Yeah - that looks like it should fix things up. What's your schedule? (I know you have more important things to work on right now).

Thanks
-Robin

commit 37297750 (2016-08-26): osc: return proper error statuses for profile loading failures

Fixes return statuses when failing out of test profiles.

Fixes #54.
closed 2016-08-26 by radhermit