analogdevicesinc/iio-oscilloscope #55: xml files are old, and outdated

rgetz, 2016-11-28:
The adi_regmap_adc.xml,  is three years old - there are updates to the core that aren't reflected in here anymore...

I think this is autogenerated:
https://wiki.analog.com/resources/fpga/docs/hdl/regmap

Noticed this when looking at 	0x0088	REG_UI_STATUS (User Interface Status) register on the ADC Common.

dNechita, 2016-11-29:
There are 2 versions of adi_regmap_adc.xml:
 - adi_regmap_adc.xml is loaded by iio-osc when a pcore major version < 9 is detected
 - adi_regmap_adc_v9.xml is loaded when a pcore major = 9

Could you please read register 0x00 (REG_VERSION) to see what is going on? The read value should start with 9.

-Dan

rgetz, 2016-11-29:
I will check when I get back to my desk. 

This was on Pluto - if you want to try before me.

dNechita, 2016-11-29:
I've checked the REG_VERSION register and it reads 0x90062, which is ok.
I'm trying to start iio-osc to see if I can reproduce the issue on my end.

dNechita, 2016-11-29:
I've connected to Pluto and opened iio-osc, but I can't reproduce the issue.  The Debug plugin shows the proper bits for register 0x0088.

If by any chance you have a newer pcore version than 9 then the Debug plugin will not load any xml file. That is something I need to fix.

dNechita, 2016-12-05:
I've added the xml file for the ADC core, version 10. Commit: https://github.com/analogdevicesinc/iio-oscilloscope/commit/15e2971d016c9a8393e200372be8b2d9700e8d6c

IIO-Oscilloscope will know which xml file to load for the core version it detects.

rgetz, 2019-03-01:
closing as outdated

closed 2019-03-01 by rgetz