analogdevicesinc/iio-oscilloscope #58: make error, cannot find -lad9361, libosc.so

mooneyj, 2017-07-28:
Fresh ubuntu 17.04 install. Latest libiio installed from repository (v0.10-6-gc95ff6a), as per instructions.

When running make -j 4, i get the following:
`  LD      libosc.so
/usr/bin/ld: cannot find -lad9361
collect2: error: ld returned 1 exit status
Makefile:107: recipe for target 'libosc.so' failed
make: *** [libosc.so] Error 1
`

lclausen-adi, 2017-07-28:
Hi,

You need to install https://github.com/analogdevicesinc/libad9361-iio

Which build instructions are you using so we can make sure that they are updated accordingly,

- Lars

mooneyj, 2017-07-28:
Hi Lars,

After installing libad9361 it now compiles! The build instructions i was referencing are here: 
https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope

I also went throught the iio-oscilloscope makefile and installed the dependencies listed there, they are similar to the ones specified in the build instructions adi_update_tools.sh. It would be easier if the user didn't have to read through the shell script in order to find the dependencies.

apt-get -y install libgtk2.0-dev libgtkdatabox-dev libmatio-dev libfftw3-dev libxml2 libxml2-dev bison flex libavahi-common-dev libavahi-client-dev libcurl4-openssl-dev libjansson-dev cmake libaio-dev

Ontop of this I installed libqtglib-2.0-0, libqt5glib-2.0-0 I'm not sure if they are actually required, but the iio-osc makefile specifies glib-2.0 as a dependency.

Thanks for your help.
John

lclausen-adi, 2017-07-28:
You only need libglib2.0-dev (no QT), but that is a dependency of some of the other packages, so it will be pull in regardless of whether you install it explicitly or not.

lclausen-adi, 2017-07-28:
Added comments about the standard package dependencies and libad9361-iio to the build instructions.

closed 2017-07-28 by lclausen-adi