analogdevicesinc/iio-oscilloscope #59: Linux Local device not found AD7195EBZ

mooneyj, 2017-07-28:
Fresh ubuntu 17.04 install onto Lenovo X1 laptop. AD7192 kernel modules enabled with modprobe. But iio-oscilloscope finds no iio devices.

Here's a screenshot showing iio-osc and dmesg outputs after plugging in the AD7195EBZ via USB.
[image]

dmesg gives a: device descriptor read/64 error -110
iio-oscilloscope gives FRU Info: No eeprom files found in /sys/

lclausen-adi, 2017-07-28:
Hi,

I'm really sorry, but connecting the EVAL board via USB is not supported since there is no driver for the USB device. Only direct connections via SPI are supported.

If you want to access the EVAL board via USB on a Linux machine using libusb and a custom application might be an option. You can find information about the USB protocol of the EVAL board here: https://ez.analog.com/thread/41703

Lars

rgetz, 2017-07-28:
The device (AD7195) would need to be connected to one of the native SPI busses. It won't work with the eval board (sorry). 

You need to stick with Windows and the eval software.

mooneyj, 2017-07-28:
That's such a shame!
Thanks for the thread Lars, i'll have a go at implementing that code.

Here's my lsusb output, incase anyone comes across this and wants to work on a usb app.

`Bus 001 Device 008: ID 0456:b483 Analog Devices, Inc. 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass       255 Vendor Specific Subclass
  bDeviceProtocol       255 Vendor Specific Protocol
  bMaxPacketSize0        64
  idVendor           0x0456 Analog Devices, Inc.
  idProduct          0xb483 
  bcdDevice            0.00
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          171
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           6
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
[... 159 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x88  EP 8 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

`

mooneyj, 2017-07-28:
Would it be possible to interface the eval board using the J2 header labelled Interface? Connecting it to the GPIO of a raspi for example.
It seems to be a serial interface though. Can the SPI interface be accessed directly on the eval board?

lclausen-adi, 2017-07-28:
Yes that should work. There is an comment about this in the [board schematics](http://www.analog.com/media/cn/technical-documentation/user-guides/UG-257.pdf#page=13)

If you run into any troubles creating the DT overlay for the RPI please open a discussion in the [EZ](https://ez.analog.com/community/linux-device-drivers/linux-software-drivers) since this is no longer directly related to the IIO scope.

mooneyj, 2017-07-28:
Thanks for your help.

closed 2017-07-28 by mooneyj