analogdevicesinc/iio-oscilloscope #60: spectrum analyzer broken?

shclift, 2017-08-02:
Configuration: zc702, fmcomms2, 2016_R2-2017_06_29.img

The spectrum analyzer window, launched from the spectrum analyzer tab by clicking the "Start Sweep" button, seems to be broken. The grid is messed up and there is no trace. Well, most of the time - sometimes something shows up (see attached image). For extra points, clicking the AutoZoom button tends to freeze the program. Running the tools update did not improve things.

It works with 2016_R1-2016_12_23.img, but the grid is still messed up.

[image]

dNechita, 2017-08-03:
Hi!

We will take a look at this and check why is not working.

trevor-crowley, 2017-08-07:
I have confirmed this issue on my Zed, AD-FMCOMMS3-EBZ, with 6 Aug update.

dNechita, 2017-08-07:
Thanks!
I've been able to reproduce this as well and found the problem. We are currently working on a solution.

dNechita, 2017-08-10:
The issue has been fixed in libiio, on which the iio-osc depends on.
https://github.com/analogdevicesinc/libiio/commit/82a23b2f94fb9dc3ddb119177e8ed11b9a4729f2
Run the adi_update_tools.sh script and things should start working again.

Regarding the grid: it never worked properly for the spectrum. I'll see what I can do about that.
Thanks!

trevor-crowley, 2017-08-10:
Works now.  Thank you.

closed 2017-08-11 by lclausen-adi