analogdevicesinc/iio-oscilloscope #61: Unable to build on OSX

borjam, 2017-08-23:
I have installed the required packages via Mac Ports and I have compiled and installed iio and ad9361. The build system has installed both as /Library/Frameworks/iio.framwork and /Library/Frameworks/ad9361.framework.

But I am unable to compile iio-oscilloscope. For a start, the Makefile has invalid options for the compiler.

Is it supposed to work on OS X? According to this page it is, but it's rather obvious that it can't be built on OS X (or I am extremely thick today!)

https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope

Thanks!

tfcollins, 2017-08-23:
This is totally possible but not with the default Makefile.  I don't think clang understands some of the MINGW variables.

I've built and ran iio-oscilloscope fine on OS 10.12 (sierra) fine though with a custom brew package, and of course a custom Makefile.

borjam, 2017-08-23:
I see, I was wondering if AD just forgot to add some files.

Maybe they'll chime in?

Thank you!

tfcollins, 2017-08-24:
No, it's just that osc is not actively built against OSX.  I know others on the team have got it to build with some changes, but that wiki page is mostly for Linux.

borjam, 2017-09-05:
Thank you :) 

I managed to do it. Quite complicated!

The source is written with Linux in mind. For now it was a quick and dirty build, I will try to do it better and contribute a set of diffs.

In short: The easy stuff was to install gcc6 from Macports, and some required dependencies like gtk, glib, fftw3...

Some surgery was needed because shared libraries have a .dylib suffic on Mac OS X. Also, Mac OS X doesn't have the memrchr() function, and some include files are in different places. Like <malloc.h> which is actually <malloc/malloc.h>.

I haven't made it work using USB for now, it's using IP. On a Macbook Air I am getting about 18 fps according to the program's standard output.

I'll keep you posted.

tfcollins, 2017-09-05:
If you have libiio and libad9361-iio you can use a brew tap:

`brew tap tfcollins/formulae`
`brew install --HEAD iio-oscilloscope`

This will get all the necessary dependencies and build everything nicely.  The changes build fine on linux but need to check on Windows before opening a PR.  Necessary changes are in my repo: https://github.com/tfcollins/iio-oscilloscope/commit/0588a7f5eb5028318b589781a23946d659dc8a90

This may work on macports, just not sure all the environmental setup that macports does during the build process.  It would be nice to get both working :)

borjam, 2017-09-05:
> On 5 Sep 2017, at 16:53, Travis Collins <notifications@github.com> wrote:
> 
> If you have libiio and libad9361-iio you can use a brew tap:
> 
> brew tap tfcollins/formulae
> brew install --HEAD iio-oscilloscope
> 
> This will get all the necessary dependencies and build everything nicely. The changes build fine on linux but need to check on Windows before opening a PR. Necessary changes are in my repo: tfcollins/iio-oscilloscope@0588a7f
> 
> This may work on macports, just not sure all the environmental setup that macports does during the build process. It would be nice to get both working :)

For now it worked :)

It reminded me the old times, compiling impossible stuff on SCO Unix! I am not developing on Linux and I was unaware of the 
pkg-config program, so I ended up installing the ports for all the dependencies and adding -I, -l and -L flags manually to the Makefile :)

I had to install gcc from Macports and I rather butchered the Makefile. Anyway this has been an ugly kludge, I’ll cleanup and try to do a clean and tidy
installation. 

One question, now that we are at it. Does the spectrum analyzer plugin work for me? When I try it I get errors setting the sample rate. 

I’ll try brew as well, thanks!

Borja.

Zaphod-2:iio-oscilloscope-0.7-master borjam$  diff Makefile Makefile.orig|more
8c8
< # FRU_FILES=$(PREFIX)/lib/fmc-tools/
---
> FRU_FILES=$(PREFIX)/lib/fmc-tools/
10c10
< CC := gcc
---
> CC := $(CROSS_COMPILE)gcc
30c30,32
< LDFLAGS=-L$(SYSROOT)/usr/lib -L/Library/Frameworks/iio.framework -L/Library/Frameworks/ad9361.framework -L/opt/local/lib -lmatio -lz -lm -lad9361 -liio -rdynamic -lxml2 -lcurl -lgtk-x11-2.0.0 -lgtkdatabox -lgdk-x11-2.0 -lglib-2.0.0 -lgdk_pixbuf-2.0.0 -ljansson -lfftw3 -lgio-2.0 -latk-1.0.0 -lpangoft2-1.0 -lgobject-2.0
---
> LDFLAGS := $(shell $(PKG_CONFIG) --libs $(DEPENDENCIES)) \
>       $(if $(WITH_MINGW),-lwinpthread) \
>       -L$(SYSROOT)/usr/lib -lmatio -lz -lm -lad9361
31a34,38
> ifeq ($(WITH_MINGW),y)
>       LDFLAGS += -Wl,--subsystem,windows
> else
>       LDFLAGS += -rdynamic
> endif
33,40c40
< CFLAGS := -I/opt/local/include/glib-2.0 -I/opt/local/lib/glib-2.0/include \
<       -I/opt/local/include/gtk-2.0 -I/opt/local/lib/gtk-2.0/include \
<       -I/opt/local/include/cairo \
<       -I/opt/local/include/pango-1.0 -I/opt/local/include/gdk-pixbuf-2.0 \
<       -I/opt/local/include/atk-1.0 \
<       -I/Library/Frameworks/iio.framework/Headers \
<       -I/Library/Frameworks/ad9361.framework/Headers \
<       -I/opt/local/include/libxml2 \
---
> CFLAGS := $(shell $(PKG_CONFIG) --cflags $(DEPENDENCIES)) \
42d41
<       -D_DARWIN_C_SOURCE \
46c45
<       -g -std=gnu90 -D_GNU_SOURCE -O2 -funwind-tables \
---
>       -Werror -g -std=gnu90 -D_GNU_SOURCE -O2 -funwind-tables \
53,54c52,59
< # SO := $(if $(WITH_MINGW),dll,so)
< SO := dylib
---
> DEBUG ?= 0
> ifeq ($(DEBUG),1)
>       CFLAGS += -DDEBUG
> else
>       CFLAGS += -DNDEBUG
> endif
> 
> SO := $(if $(WITH_MINGW),dll,so)
79,80c84,85
<       plugins/spectrum_analyzer.$(SO) \
<       plugins/scpi.$(SO)
---
>       $(if $(WITH_MINGW),,plugins/spectrum_analyzer.so) \
>       $(if $(WITH_MINGW),,plugins/scpi.so)
118c123
<       $(CMD)$(CC) $(CFLAGS) $< $(LDFLAGS) -L. -losc -shared memrchr.o -o $@
---
>       $(CMD)$(CC) $(CFLAGS) $< $(LDFLAGS) -L. -losc -shared -o $@
Zaphod-2:iio-oscilloscope-0.7-master borjam$

borjam, 2017-09-05:
> On 5 Sep 2017, at 16:53, Travis Collins <notifications@github.com> wrote:
>
> If you have libiio and libad9361-iio you can use a brew tap:
>
> brew tap tfcollins/formulae
> brew install --HEAD [...]

I have done a quick test and it seems it’s not finding the plugins. Am I missing anything?

(Note that I followed your recipe blindly)

Borja.

tfcollins, 2017-09-05:
At the moment I don't include the spectrum analyzer and scpi plugins.  They will need more specific changes to the source.  MINGW builds don't include them as well.

tfcollins, 2017-09-06:
@borjam I enabled the spectrum analyzer plugin in my repo.  It should work ok.  Working through the scpi plugin.

borjam, 2017-09-06:
> On 6 Sep 2017, at 04:58, Travis Collins <notifications@github.com> wrote:
> 
> @borjam I enabled the spectrum analyzer plugin in my repo. It should work ok. Working through the scpi plugin.

Sorry to insist, one of us is missing something :)

Have you tried it? What I observed when running it myself was that the program didn’t actually work. It creates
several shared libraries that get installed on /usr/local/lib/osc. Turns out that the supplied Makefile creates the
files with a .so extension and the calls to the dynamic linker don’t seem to understand them. Those libraries
are called “plugins” by the authors and they are needed to actually use the program. For example if you 
want to open a spectrum window and be able to change the frequency, bandwidth, sample rate, etc.

What I did was to modify the Makefile and the osc.c. The dynamic libraries are now generated with a .dylib
extension and the osc.c program searches for .dylib files to be passed to the dynamic linker.

The result: the main window from your brew build is blank. Compare to the one in my build. Mine is built
with the X11 version of gtk but that should be irrelevant. I just had problems to install the Quartz version of 
gtk.

I’ll see if I can cleanup the mess I made (or, better said, do it again but properly) and contribute Mac OS X compatible
changes to the Makefile and some relevant files.

rgetz, 2017-09-06:
I don't think the spectrum analyzer ever worked from a remote standpoint.

@dNechita would know from sure.

-Robin

tfcollins, 2017-09-06:
Spectrum analyzer doesn't work on linux remote, so yeah I'm guessing it never did.

@borjam Just use the frequency plot if you want a similar measurement.

borjam, 2017-09-06:
> On 6 Sep 2017, at 17:35, Travis Collins <notifications@github.com> wrote:
> 
> Spectrum analyzer doesn't work on linux remote, so yeah I'm guessing it never did.
> 
> @borjam Just use the frequency plot if you want a similar measurement.

Thank you :)

The rest seems to work fine. 

Borja.

dNechita, 2017-10-03:
Indeed, the iio-oscilloscope  spectrum analyzer plugin, when connected to a remote target is not very useful. I don't think it crashes or anything but it is unusable due to timing issues. The best environment for the spectrum analyzer plugin is when iio-osc runs directly on the target.

Dan

commodo, 2018-10-04:
This should be resolved with the current cmake integration.
Essentially, the build steps included in these 2 scripts should get a working build on OS X.
https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/CI/travis/before_install_darwin
https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/CI/travis/make

Will close this.
If there are still issues or open items on this, we can re-open and continue here.

Thanks
Alex

closed 2018-10-04 by commodo