analogdevicesinc/iio-oscilloscope #64: Integrate filter designer from libad9361

tfcollins, 2018-01-12:
Example filter generation: https://github.com/analogdevicesinc/libad9361-iio/blob/filter-designer-standalone/test/filter_designer_test.c

This will supplement and/or replace the current baseband_auto_rate function which uses a few pre-designed filters.  However, extra parameters will need to be exposed in OSC for this designer.

mhennerich, 2018-01-15:
Did we come to an conclusion which parameters we must expose?

tfcollins, 2018-01-15:
I think exposing:
Apass
Astop
Fpass
Fstop

would work.  The remaining parameters can be calculated or pulled from existing parameters in the tool.

rgetz, 2018-01-15:
Did we validate things where Fpass == Fstop == Fs/2?
or
Fpass = 90% of Fs/2 and Fstop == Fs/2?