analogdevicesinc/iio-oscilloscope #65: Only runs in directory containing glade files

darsor, 2018-02-05:
I am trying to install this on Arch Linux with the install `PREFIX=/usr` instead of `PREFIX=/usr/local`.

When I install it with the default prefix, `/usr/local`, it works fine with no errors or warnings. (I do have to set `LD_LIBRARY_PATH=/usr/local/lib`).

When I install it to the custom prefix `/usr` and run it, no windows pop up, and I get the following errors:

```
(osc:8632): Gtk-CRITICAL **: IA__gtk_about_dialog_set_version: assertion 'GTK_IS_ABOUT_DIALOG (about)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_label_get_label: assertion 'GTK_IS_LABEL (label)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_label_set_label: assertion 'GTK_IS_LABEL (label)' failed

(osc:8632): Gtk-CRITICAL **: gtk_box_pack: assertion 'GTK_IS_BOX (box)' failed

(osc:8632): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(osc:8632): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(osc:8632): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(osc:8632): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(osc:8632): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(osc:8632): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(osc:8632): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(osc:8632): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(osc:8632): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(osc:8632): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(osc:8632): GLib-GObject-CRITICAL **: g_object_bind_property_full: assertion 'G_IS_OBJECT (source)' failed

(osc:8632): GLib-GObject-CRITICAL **: g_object_bind_property_full: assertion 'G_IS_OBJECT (source)' failed

(osc:8632): GLib-GObject-CRITICAL **: g_object_bind_property_full: assertion 'G_IS_OBJECT (source)' failed

(osc:8632): GLib-GObject-CRITICAL **: g_object_bind_property_full: assertion 'G_IS_OBJECT (source)' failed

(osc:8632): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(osc:8632): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed

(osc:8632): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(osc:8632): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed
ERROR: Bad URI: 'usb:expected_error'

(osc:8632): Gtk-CRITICAL **: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_toggle_button_set_active: assertion 'GTK_IS_TOGGLE_BUTTON (toggle_button)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_combo_box_get_model: assertion 'GTK_IS_COMBO_BOX (combo_box)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_list_store_clear: assertion 'GTK_IS_LIST_STORE (list_store)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_combo_box_text_append_text: assertion 'GTK_IS_COMBO_BOX_TEXT (combo_box)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_combo_box_set_active: assertion 'GTK_IS_COMBO_BOX (combo_box)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed

(osc:8632): GLib-GObject-WARNING **: invalid (NULL) pointer instance

(osc:8632): GLib-GObject-CRITICAL **: g_signal_connect_data: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_widget_show: assertion 'GTK_IS_WIDGET (widget)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_check_menu_item_set_active: assertion 'GTK_IS_CHECK_MENU_ITEM (check_menu_item)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_check_menu_item_set_active: assertion 'GTK_IS_CHECK_MENU_ITEM (check_menu_item)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_window_move: assertion 'GTK_IS_WINDOW (window)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_widget_set_sensitive: assertion 'GTK_IS_WIDGET (widget)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_dialog_run: assertion 'GTK_IS_DIALOG (dialog)' failed
unknown response (-1) in dialogs.c(fru_connect_dialog)

(osc:8632): Gtk-CRITICAL **: IA__gtk_widget_hide: assertion 'GTK_IS_WIDGET (widget)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_check_menu_item_get_active: assertion 'GTK_IS_CHECK_MENU_ITEM (check_menu_item)' failed
^C
(osc:8632): Gtk-CRITICAL **: IA__gtk_notebook_get_n_pages: assertion 'GTK_IS_NOTEBOOK (notebook)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_window_get_position: assertion 'GTK_IS_WINDOW (window)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_check_menu_item_get_active: assertion 'GTK_IS_CHECK_MENU_ITEM (check_menu_item)' failed

(osc:8632): Gtk-CRITICAL **: IA__gtk_check_menu_item_get_active: assertion 'GTK_IS_CHECK_MENU_ITEM (check_menu_item)' failed
```

However, when I run the installed executable from any directory containing the glade files, it does seem to work correctly, with the warnings

```
(osc:9627): Gtk-WARNING **: Could not load image 'osc128.png': Failed to open file '/home/darsor/build/iio-oscilloscope-git/src/iio-oscilloscope/osc128.png': No such file or directory

(osc:9627): Gtk-WARNING **: Could not load image 'IIOlogo.png': Failed to open file '/home/darsor/build/iio-oscilloscope-git/src/iio-oscilloscope/IIOlogo.png': No such file or directory

(osc:9627): Gtk-WARNING **: Could not load image 'ADIlogo.png': Failed to open file '/home/darsor/build/iio-oscilloscope-git/src/iio-oscilloscope/ADIlogo.png': No such file or directory
```

It seems like something in the source or executable does not support the custom prefix.

lclausen-adi, 2018-02-05:
Hi,

This is supposed to work, as long as PREFIX is set to the same value during both the build and the install phase. See https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/config.h

darsor, 2018-02-06:
That did it, thanks for the quick response!

closed 2018-02-06 by darsor