analogdevicesinc/iio-oscilloscope #68: "Unable to claim interface"

cpbridges, 2018-02-22:
Hi AD,

I've built and install the ADI IIO Oscilloscope application from git/master on Ubuntu 16.04 but have  an issue in opening the FFT plot. Note that it would appear that RSSI is coming through but parts of the scope are clearly not working. I've screenshotted it below, and how it runs from the terminal.

[image]

Any ideas how to get the fft up?

Best, Chris

closed 2018-02-22 by cpbridges

mhennerich, 2018-02-22:
File->New Plot ?

cpbridges, 2018-02-22:
I was totally clicking on the wrong thing in the interface, like an idiot. I think I did the same thing about 10 months ago. Best, Chris