analogdevicesinc/iio-oscilloscope #71: Incomplete function declaration scpi_rx_setup()

UweBonnes, 2018-03-17:
Hello,

compiling master gives:
  LD      plugins/scpi.so
plugins/scpi.c: In function ‘scpi_handle’:
plugins/scpi.c:984:4: error: call to function ‘scpi_rx_setup’ without a real prototype [-Werror=unprototyped-calls]
    scpi_rx_setup(current_instrument);
    ^
plugins/scpi.c:601:6: note: ‘scpi_rx_setup’ was declared here
 void scpi_rx_setup()
      ^
cc1: all warnings being treated as errors
Makefile:140: recipe for target 'plugins/scpi.so' failed

I can fix e.g. by
diff --git a/plugins/scpi.c b/plugins/scpi.c
index f7ea719..bb1a547 100644
--- a/plugins/scpi.c
+++ b/plugins/scpi.c
@@ -598,7 +598,7 @@ void scpi_rx_set_bandwith_auto(double ratio)
 
 }
 
-void scpi_rx_setup()
+void scpi_rx_setup(void *bla)
 {
        static time_t rx_cal_time = 0;

Bye

rgetz, 2018-03-17:
Uwe:

What toolchain are you using? 

gcc will warn on your patch due to -Wunused-parameter , which is set in -Wextra. 

?

-Robin

UweBonnes, 2018-03-17:
I use OpenSuse Leap 42.3. The default GCC is gcc version is 4.8.5 (SUSE Linux). However "void scpi_rx_setup();" is no valid C code as far as I know.

UweBonnes, 2018-03-17:
The diff I gave above was only to menat as a hint. A more complete solution would be:
diff --git a/plugins/scpi.c b/plugins/scpi.c
index f7ea719..6455611 100644
--- a/plugins/scpi.c
+++ b/plugins/scpi.c
@@ -598,10 +598,10 @@ void scpi_rx_set_bandwith_auto(double ratio)
 
 }
 
-void scpi_rx_setup()
+void scpi_rx_setup(struct scpi_instrument *instrument)
 {
        static time_t rx_cal_time = 0;
-
+        (void)instrument;
        scpi_fprintf(&spectrum_analyzer, ":DISP:TRACE:Y:RLEVEL %d DBM\n", 10);
        /* Turn averaging off */
        scpi_fprintf(&spectrum_analyzer, ":AVER OFF\n");

rgetz, 2018-03-17:
When you declare a function with an empty argument list, it invokes K&R (pre-prototype) semantics and nothing is assumed about the parameter list; this is so that pre-ANSI C code will still compile.

You are correct - in that proper ANSI-C prototype declaration for a function with an empty parameter list should use (void) instead of ().

I will update.  Thanks for showing how old & sloppy am. :) I should go through and fix all the -pedantic errors, but it has always been low on the list. (Any volunteers?)

commodo, 2018-03-29:
@UweBonnes 
[I'm curious] are you (by any chance) using a non-x86 OpenSUSE ?
AFAICT, there are some unofficial non-x86 ports to ppc64le and aarch64 here: https://software.opensuse.org/distributions/leap

In the past, I've noticed that GCC compilers for PPC are a bit more picky.
Not that they are incorrect, but they complain a bit more compared to x86 versions (and some other archs) of GCC.

UweBonnes, 2018-04-01:
No, it is Plain OpenSuse 42.3 on X64

commodo, 2018-04-01:
ack
maybe there are some extra build options that OpenSUSE adds for it's version of GCC
thanks for the clarification 

would it be too much to ask you to also test PR #78  ? :)

thanks
Alex

UweBonnes, 2018-04-02:
The patch fixes the "without a real prototype " error.

closed 2018-04-27 by commodo