analogdevicesinc/iio-oscilloscope #72: LDFLAGS:  -L$(SYSROOT)/usr/lib vs  -L$(SYSROOT)/usr/lib64

UweBonnes, 2018-03-17:
Hello,

compiling on Opensuse Leap42.3 gives a lot of errors like
  LD      libosc.so
/usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../x86_64-suse-linux/bin/ld: inkompatibles /usr/lib/libfontconfig.so wird bei der Suche nach -lfontconfig übersprungen
/usr/lib64/gcc/x86_64-suse-linux/4.8/../../../../x86_64-suse-linux/bin/ld: inkompatibles /usr/lib/libfreetype.so wird bei der Suche nach -lfreetype übersprungen

Changing
diff --git a/Makefile b/Makefile
index 418af2e..e08900f 100644
--- a/Makefile
+++ b/Makefile
@@ -38,7 +38,7 @@ $(foreach dep,$(DEPENDENCIES),$(eval $(call dep_flags,$(dep))))
 
 LDFLAGS := $(DEP_LDFLAGS) \
        $(if $(WITH_MINGW),-lwinpthread) \
-       -L$(SYSROOT)/usr/lib -lmatio -lz -lm -lad9361
+       -L$(SYSROOT)/usr/lib64 -lmatio -lz -lm -lad9361
 
 ifeq ($(WITH_MINGW),y)
        LDFLAGS += -Wl,--subsystem,windows

makes the error go away for me, but a more general solution is needed.

commodo, 2018-03-28:
One generic way to fix this, is to add `-L$(SYSROOT)/usr/lib64` before `-L$(SYSROOT)/usr/lib`.
It's not perfect, but it should handle most cases.
If the lib is found in `-L$(SYSROOT)/usr/lib64`, it will be used.
If not, and is found in ``-L$(SYSROOT)/usr/lib`, it will be used.

It's also a simple solution to the problem.
I'm not sure about any other solution here.

I'm open to suggestions.
Otherwise, I propose we solve it like this.

I will spin up a PR with this change.

commodo, 2018-03-28:
@UweBonnes
Could you test the change in PR #77 ?
If you give the ok, we could then merge it.

Thanks :)
Alex

UweBonnes, 2018-04-01:
PR#77 fixes the problem on OpenSuse 42,3, X64

closed 2018-04-18 by commodo