analogdevicesinc/iio-oscilloscope #73: Spectrum analyser: Failed to set the rx sampling rate to 61,440000

UweBonnes, 2018-03-17:
Hello,

starting the spectrum analyser compiled fron git master on OpenSuse Leap 42.3 gave:
 Failed to set the rx sampling rate to 61,440000in configure_data_capture

Instrumenting the code shows that tha actual rate is only off a little:
Failed to set the rx sampling rate to 61,440000in configure_data_capture, out 61,439999 

What about aloowing some tolerance?
diff --git a/plugins/spectrum_analyzer.c b/plugins/spectrum_analyzer.c
index ad6f2d1..da7fc4b 100644
--- a/plugins/spectrum_analyzer.c
+++ b/plugins/spectrum_analyzer.c
@@ -398,7 +398,8 @@ static bool configure_data_capture(plugin_setup *setup)
                dev_info->adc_scale = '?';
                dev_info->adc_freq = 0.0;
        }
-       if (dev_info->adc_freq != sampling_rate) {
+       if ((dev_info->adc_freq > (1.01 * sampling_rate)) ||
+            (dev_info->adc_freq < (0.99 * sampling_rate))) {
                fprintf(stderr, "Failed to set the rx sampling rate to %f"
                        "in %s\n", sampling_rate, __func__);
                return false;

UweBonnes, 2018-04-29:
Well, the problem may stop first time users. You know, the span of attention for most people is quit small.

hakeen980, 2019-05-21:
Hi Commodo, 

do you have an udoate in meantime? 
I get the same error under Ubuntu 18.04.2.
With a relatively older ADALM-PLUTO

UweBonnes, 2019-05-22:
hakeen980 writes:
> Hi Commodo,
> 
> do you have an udoate in meantime?
> I get the same error under Ubuntu 18.04.2.
> With a relatively older ADALM-PLUTO

It's been a long time when I played with the Pluto. I think I got
around that problem by hacking the code...
-- 
Uwe Bonnes                bon@elektron.ikp.physik.tu-darmstadt.de

Institut fuer Kernphysik  Schlossgartenstrasse 9  64289 Darmstadt
--------- Tel. 06151 1623569 ------- Fax. 06151 1623305 ---------

commodo, 2019-05-24:
can you try PR https://github.com/analogdevicesinc/iio-oscilloscope/pull/161 ?

commodo, 2019-06-05:
PR was merged.
If there are any issues, feel free to re-open.

closed 2019-06-05 by commodo