analogdevicesinc/iio-oscilloscope #74: Make always recompiles all files

UweBonnes, 2018-03-17:
Hello,

it seems the Makefile or dependancies have  some flaw. "make" always recompiles all files. This makes test, debug, recompile, test sessions more tedious.

Bye

mhennerich, 2018-03-19:
Hmm - this was introduced with b78579ad7f5d34f1d9f035f92d22a068c747fb20

if you remove check_deps from the object file dependencies things start to work again:

@commodo can you fix this? 

# Dependencies
osc.o:  iio_widget.h int_fft.h osc_plugin.h osc.h libini2.h
oscmain.o:  config.h osc.h
oscplot.o:  oscplot.h osc.h datatypes.h iio_widget.h libini2.h
datatypes.o:  datatypes.h
iio_widget.o:  iio_widget.h
fru.o:  fru.h
dialogs.o:  fru.h osc.h
trigger_dialog.o:  fru.h osc.h iio_widget.h
xml_utils.o:  xml_utils.h
phone_home.o:  phone_home.h
plugins/dac_data_manager.o:  plugins/dac_data_manager.h

commodo, 2018-03-19:
yep ; i'll take care of it;
i already started looking into it a bit ;

was wondering if it makes sense to do a conversion to CMake altogether ; 
but i'll see;