analogdevicesinc/iio-oscilloscope #79: Crash on attempt to connect  to remote

ekigwana, 2018-03-30:
master branch.
I have no local devices so Remote Devices (network) is automatically selected. I enter a hostname, click refresh and the Context infor box and IIO Devices box is populated.

When I click OK the program crashes after printing "ERROR: Bad URI: 'usb:expected_error'" to the console.

Here is the backtrace:
```
Thread 1 "osc" received signal SIGSEGV, Segmentation fault.
0x00007ffff35279a7 in __readdir (dirp=dirp@entry=0x0) at ../sysdeps/posix/readdir.c:44
44        __libc_lock_lock (dirp->lock);
(gdb) bt
#0  0x00007ffff35279a7 in __readdir (dirp=dirp@entry=0x0) at ../sysdeps/posix/readdir.c:44
#1  0x00007ffff7ba57b8 in load_plugins (notebook=0x5555558de330, ini_fn=0x0) at osc.c:898
#2  0x00007ffff7ba6243 in do_init (new_ctx=new_ctx@entry=0x555555be81e0) at osc.c:1817
#3  0x00007ffff7ba63c9 in application_reload (new_ctx=0x555555be81e0, load_profile=<optimized out>) at osc.c:1525
#4  0x00007ffff7bbf6dd in fru_connect_dialog (data=0x7ffff7dd7800 <dialogs>, load_profile=<optimized out>) at dialogs.c:677
#5  0x0000555555556cb1 in main (argc=<optimized out>, argv=<optimized out>) at oscmain.c:250
```

ekigwana, 2018-03-30:
So I got to the bottom of the issue. I have a multilib environment that does not allow installing 64 bit libraries to /usr/lib using the package manager. Turns out that osc is hard coded to look for the plugin dir in /usr/lib so when my package manager installed to /usr/lib64 osc could no longer locate the plugins.

I recommend: 
1 - Checking for a plugin directory and reporting if it is missing.
2 - Check /usr/lib64 /usr/lib /usr/lib32 before giving up.

commodo, 2018-03-31:
Hey,

Thanks for the report and looking into it.

Regarding point 2, there is PR #77 that tries to handle this, since something similar was reported via #72 
It just adds /usr/lib64 to the lib search path when linking. 
I'm not sure about adding `/usr/lib32` ; I personally did not see it in any distros/OSes.
But I guess it could be fine to add it.

I'm wondering if PR #77 is sufficient to fix this for you.
Could you test it please ? :)

closed 2018-04-18 by commodo