analogdevicesinc/iio-oscilloscope #85: Build information

AtharBaig0027, 2018-06-01:
hi, 

I'm looking for build information for iio-oscilloscope and it's plug-ins. I checked the wiki page of this thread but build information is missing. 

Please help me find build information.

mhennerich, 2018-06-01:
Please see here: https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope

mhennerich, 2018-06-05:
Fixed with this commit: https://github.com/analogdevicesinc/iio-oscilloscope/commit/c89c6c73a2366d96b9980b44f2c58dcd1c98a25d

Can you try again? what version of GCC are you using?

-Michael

AtharBaig0027, 2018-06-05:
I removed the unused constant and it built successfully. Thanks. 
I'm using gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0.

commodo, 2018-10-04:
This looks resolved.
Will close.

If there are still open issues we can re-open or continue here.

Thanks
Alex

closed 2018-10-04 by commodo