analogdevicesinc/iio-oscilloscope #93: Build error

Skizzo69, 2018-08-06:
[skizzo@localhost iio-oscilloscope]$ make
  CC      oscmain.o
  CC      osc.o
osc.c: In function ‘capture_start’:
osc.c:1325:70: error: cast between incompatible function types from ‘gboolean (*)(void)’ {aka ‘int (*)(void)’} to ‘gboolean (*)(void *)’ {aka ‘int (*)(void *)’} [-Werror=cast-function-type]
   capture_function = g_timeout_add_full(G_PRIORITY_DEFAULT_IDLE, 50, (GSourceFunc) capture_process, NULL, NULL);
                                                                      ^
cc1: all warnings being treated as errors
make: *** [Makefile:145: osc.o] Error 1

commodo, 2018-08-07:
hmm;
what gcc version are you using ?
and which distro ?

these seem like reasonable errors; but i'm not getting them yet;
this stuff typically appears when moving to a newer GCC

lclausen-adi, 2018-08-07:
This is gcc 8. I have fixes, will push them later.

commodo, 2018-08-07:
Ah, cool :)
Reduces double-work.

Thanks

lclausen-adi, 2018-08-07:
Hi @Skizzo69, the errors should be fixed now in the latest master branch.

Skizzo69, 2018-08-07:
Fixed those errors. Now I'm getting errors related to libini2.c? Should I close this and open another issue then?

lclausen-adi, 2018-08-07:
What you can do for now, in the Makefile remove "-Werror" and let us know all the warnings you get when building the osc.

Skizzo69, 2018-08-07:
This is what I'm getting now.

[skizzo@localhost iio-oscilloscope]$ make -j 4
  CC      osc.o
  CC      oscmain.o
  CC      datatypes.o
  CC      oscplot.o
  CC      int_fft.o
  CC      iio_widget.o
  CC      fru.o
  CC      dialogs.o
  CC      trigger_dialog.o
  CC      xml_utils.o
  CC      libini/libini.o
  CC      libini2.o
  CC      phone_home.o
  CC      plugins/dac_data_manager.o
libini2.c: In function ‘loop_expand’:
libini2.c:485:45: error: ‘>’ directive output may be truncated writing 1 byte into a region of size between 0 and 127 [-Werror=format-truncation=]
  snprintf(loop->var, sizeof(loop->var), "<%s>", var);
                                             ^
libini2.c:485:2: note: ‘snprintf’ output between 3 and 130 bytes into a destination of size 128
  snprintf(loop->var, sizeof(loop->var), "<%s>", var);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:144: libini2.o] Error 1
make: *** Waiting for unfinished jobs....
  CC      plugins/fir_filter.o
[skizzo@localhost iio-oscilloscope]$

[skizzo@localhost iio-oscilloscope]$ sudo make install
[sudo] password for skizzo: 
  CC      libini2.o
libini2.c: In function ‘loop_expand’:
libini2.c:485:45: error: ‘>’ directive output may be truncated writing 1 byte into a region of size between 0 and 127 [-Werror=format-truncation=]
  snprintf(loop->var, sizeof(loop->var), "<%s>", var);
                                             ^
libini2.c:485:2: note: ‘snprintf’ output between 3 and 130 bytes into a destination of size 128
  snprintf(loop->var, sizeof(loop->var), "<%s>", var);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:145: libini2.o] Error 1

madengr, 2018-09-18:
I'm getting the same build error with gcc 8.1.1. on Fedora 28. 

Any solution?

commodo, 2018-09-24:
We're waiting for a re-spin of that PR based on comments.

If it won't happen [in a week or so] we'll see about implementing this ourselves.

SarahWicker, 2019-02-27:
Hey, 
I don't know if this is the right place to write it but i also have a build Problem on the Building of iio-oscilloscope :
`root@linaro-ubuntu-desktop:/home/linaro/iio-oscilloscope# make -j 4
[  2%] [  5%] [  8%] [ 10%] Building C object CMakeFiles/osc.dir/oscplot.c.o
Building C object CMakeFiles/osc.dir/trigger_dialog.c.o
Building C object CMakeFiles/osc.dir/dialogs.c.o
Building C object CMakeFiles/osc.dir/xml_utils.c.o
[ 13%] Building C object CMakeFiles/osc.dir/libini/libini.c.o
[ 16%] Building C object CMakeFiles/osc.dir/libini2.c.o
[ 18%] Building C object CMakeFiles/osc.dir/phone_home.c.o
[ 21%] /home/linaro/iio-oscilloscope/oscplot.c:403:37: error: unknown type name âosc_plotâ
/home/linaro/iio-oscilloscope/oscplot.c:403:47: error: expected declaration specifiers or â...â before â(â token
/home/linaro/iio-oscilloscope/oscplot.c: In function âosc_plot_initâ:
/home/linaro/iio-oscilloscope/oscplot.c:442:2: error: implicit declaration of function âosc_plot_get_instance_privateâ [-Werror=implicit-function-declaration]
/home/linaro/iio-oscilloscope/oscplot.c:442:13: error: assignment makes pointer from integer without a cast [-Werror]
/home/linaro/iio-oscilloscope/oscplot.c: In function âosc_plot_disposeâ:
/home/linaro/iio-oscilloscope/oscplot.c:838:2: error: âosc_plot_parent_classâ undeclared (first use in this function)
/home/linaro/iio-oscilloscope/oscplot.c:838:2: note: each undeclared identifier is reported only once for each function it appears in
/home/linaro/iio-oscilloscope/oscplot.c: In function âosc_plot_finalizeâ:
/home/linaro/iio-oscilloscope/oscplot.c:843:2: error: âosc_plot_parent_classâ undeclared (first use in this function)
Building C object CMakeFiles/osc.dir/plugins/dac_data_manager.c.o
/home/linaro/iio-oscilloscope/oscplot.c: At top level:
/home/linaro/iio-oscilloscope/oscplot.c:841:13: error: âosc_plot_finalizeâ defined but not used [-Werror=unused-function]
/home/linaro/iio-oscilloscope/oscplot.c:836:13: error: âosc_plot_disposeâ defined but not used [-Werror=unused-function]
/home/linaro/iio-oscilloscope/oscplot.c:440:13: error: âosc_plot_initâ defined but not used [-Werror=unused-function]
[ 24%] Building C object CMakeFiles/osc.dir/plugins/fir_filter.c.o
cc1: all warnings being treated as errors
[ 27%] make[2]: *** [CMakeFiles/osc.dir/oscplot.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
Building C object CMakeFiles/osc.dir/eeprom.c.o
make[1]: *** [CMakeFiles/osc.dir/all] Error 2
make: *** [all] Error 2 `

I am Building on a Linaro distribution ( Linaro 12.11)
I was able to build libio and the libad9361 without Problem. 
If anyone has an idea of why that would be great :)
thank you a lot for you help 
Sarah

dNechita, 2019-02-27:
Hi Sarah,

Yes, this is the right place, but can you please open a new issue and re-post everything in that issue?
And please put the error log inside a file and attach it. It is much easier to read that way.

Thanks,
Dan

rgetz, 2019-03-01:
Since #103 was marged - and fixes things - I will close this. If you run into different build issues - please report a new error.

closed 2019-03-01 by rgetz