analogdevicesinc/iio-oscilloscope #102: GKSu deprecated

darsor, 2018-09-04:
The `adi-osc.desktop` file included with the installation uses `gksudo` to launch the program. Unfortunately, GKSu has been deprecated and is starting to be dropped in several distributions, such as [Ubuntu 18.04](https://itsfoss.com/gksu-replacement-ubuntu/) and [Archlinux](https://lists.archlinux.org/pipermail/arch-dev-public/2018-April/029224.html).

As a result, the desktop launcher no longer works. Would it be difficult to switch to PolicyKit?

commodo, 2018-09-05:
Will take a look.
Thanks for reporting.

I don't know how difficult it would be to switch to PolicyKit.
But we'll see.

mika, 2018-11-30:
Should be enough to just invoke `pkexec /usr/bin/osc` to get this working with PolicyKit. HTH

darsor, 2018-12-21:
It looks like there's some configuration needed. When I run `pkexec /usr/bin/osc` I get the following:
```
==== AUTHENTICATING FOR org.freedesktop.policykit.exec ====
Authentication is needed to run `/usr/bin/osc' as the super user

polkit-agent-helper-1: error response to PolicyKit daemon: GDBus.Error:org.freedesktop.PolicyKit1.Error.Failed: No session for cookie
==== AUTHENTICATION FAILED ====
Error executing command as another user: Not authorized
This incident has been reported
```
Why does it need to run as root in the first place? Seems like a better solution would be to eliminate or bypass the need to run the graphical application as root.

mika, 2019-01-31:
@darsor - ah, a policy file in `/usr/share/polkit-1/actions/`, similar to e.g. https://sources.debian.org/src/nmap/7.70+dfsg1-6/debian/com.ubuntu.zenmap.policy/ might be needed then as well.

closed 2019-04-15 by commodo

commodo, 2019-04-16:
PR #156 should have fixed this.

PR #156 (2019-04-15): Fix gksu deprecated
This PR aims to fix #102. It also configures and installs an adi-osc.desktop file when using cmake. Also note that, the polkit action is only being supported on cmake builds....