analogdevicesinc/iio-oscilloscope #109: Building requires git repo

tfcollins, 2018-10-01:
Users downloading the zip will get build failures since we directly reference git variables.

Here are some (might be others)
https://github.com/analogdevicesinc/iio-oscilloscope/blob/master/dialogs.c#L906
https://github.com/analogdevicesinc/iio-oscilloscope/blob/c37d0c2fa1e68dfdb301b4fef6cdc8974c91f102/dialogs.c#L946

JanVerlinden, 2018-10-04:
The bug also exist if you use the Makefile or the CMakeLists.txt.

Building C object CMakeFiles/osc.dir/dialogs.c.o
/home/zynqdev/libiio-eclipse/osc-project/dialogs.c: In function ‘date_compare_against_build_date’:
/home/zynqdev/libiio-eclipse/osc-project/dialogs.c:906:21: error: too few arguments to function ‘atol’
  glong build_time = atol(GIT_COMMIT_TIMESTAMP);

commodo, 2018-10-04:
Yep.
We're addressing both cases.

PR #110 (2018-10-12): build: handle non-git repo builds
Fixes #109
closed 2018-10-12 by commodo