analogdevicesinc/iio-oscilloscope #111: OSC fails to build on our SD card ubuntu image

mhennerich, 2018-10-08:
adi_update_tools fails to build OSC.

root@analog:/usr/local/src/iio-oscilloscope# make
  CC      oscmain.o
  CC      osc.o
  CC      oscplot.o
  CC      datatypes.o
  CC      iio_widget.o
  CC      fru.o
  CC      dialogs.o
dialogs.c:28:64: fatal error: libserialport.h: No such file or directory
 #include <libserialport.h> /* cross platform serial port lib */
                                                                ^
compilation terminated.
make: *** [dialogs.o] Error 1
root@analog:/usr/local/src/iio-oscilloscope# cmake .
CMake Error at CMakeLists.txt:18 (cmake_minimum_required):
  CMake 3.0.0 or higher is required.  You are running version 2.8.12.2

-- Configuring incomplete, errors occurred!
root@analog:/usr/local/src/iio-oscilloscope# 

root@analog:/usr/local/src/iio-oscilloscope# cat /etc/issue
Linaro 14.04 \n \l

root@analog:/usr/local/src/iio-oscilloscope# git status
On branch master
Your branch is up-to-date with 'origin/master'.

tfcollins, 2018-10-08:
Are we still supporting the Makefile or are we switching to CMake?

mhennerich, 2018-10-08:
We should switch to cmake. However the version must be supported by the image...

closed 2018-10-15 by commodo