analogdevicesinc/iio-oscilloscope #125: Crashes on Windows

tfcollins, 2019-02-08:
With the latest OSC (v0.9) we have been seeing consistent crashes on Windows platforms when modifying parameters when scopes are running. Primarily the constellation scope.

Here are some repeatable steps that make OSC crash:
[IIO_Lab1.pdf](https://github.com/analogdevicesinc/iio-oscilloscope/files/2845867/IIO_Lab1.pdf)

This has been true for Win7 and Win10.

mhennerich, 2019-02-08:
@adisuciu 
I know you fixed this. Are we ready to create a new release?

mhennerich, 2019-03-26:
We have released a new version which should address these issues:
https://github.com/analogdevicesinc/iio-oscilloscope/releases/tag/v0.10-master

Please test and close

tfcollins, 2019-03-26:
Seems to be working better now. Still need to sign exe release.

closed 2019-11-04 by tfcollins