analogdevicesinc/iio-oscilloscope #135: how to make it work under ubuntu

Tinkimi, 2019-02-27:
Hello,i am a student learning this knowledge,and i had an newly ubuntu system ,but i don't know how to make this code to work,i wish if you could help me！ Thanks a lot!

dNechita, 2019-02-27:
Hi,

You can find build instructions at:
https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope

Tinkimi, 2019-02-27:
Pretty thanks first!
And I had done the steps mentioned at that website,but i don't know how to make this code run like an opening an dsw with vc++ ,so what I want to know is under what condition can I open the project not a single code  and make it run. 
Really thanks!

Tinkimi, 2019-02-27:
> Hi,
> 
> You can find build instructions at:
> https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope

Well, I had run this project once tonight,but now when I restart my ubuntu system,I can't run it at all,and now it's shown the error like this "./osc: error while loading shared libraries: libosc.so: cannot open shared object file: No such file or directory", I wonder if you could help me how could I solve this program,really thanks!

dNechita, 2019-02-27:
This repository provides the source code and the build instructions which are contained by the Makefile (on Linux). This is all that is required to build and run the software. You can use any IDE which has an editor and allows you to automate the steps you've seen and done in the doc I've send you. For example you could use Visual Code, Eclipse, Kate, Codeblocks and so on.

Regarding your 'cannot open shared object file' issue, please take a look at step 6 of the documentation I've sent you(https://wiki.analog.com/resources/tools-software/linux-software/iio_oscilloscope#linux). If you didn't choose to install osc then you need do this step each time you open up a new shell to run osc. If you install osc then should work out-of-the-box.

dNechita, 2019-03-11:
@Tinkimi did you manage to make iio-oscilloscope work under ubuntu?

Tinkimi, 2019-03-19:
@dNechita 　My friends, I think I should appologize for this unpoliteness action first.I am pretty sorry for long days no thanks for your help.   That day afternoon  with you and the googler's help ,I've solved this porblem,and now it runs well.Thanks very much,

dNechita, 2019-03-19:
@Tinkimi don't worry about it. We're glad you managed to make things work.

closed 2019-03-19 by dNechita