analogdevicesinc/iio-oscilloscope #136: Make error : in oscplot parameter not recognized

SarahWicker, 2019-02-28:
Hey Everyone,
I am trying to build iio-oscilloscope on my ZedBoard (using Linaro 12.11 Distribution) and when i try to build it using make -j 4 i get error messages on the oscplot file. (log message in the picture attached) Itz seems not to recognized parameters. 

I though it could come from the librairies but i previously build the library libiio and libad9361 without any Problem and have done on both the sudo make install

Version for cmake is 2.8.7 and using GNU make 3.81
Does someone has an idea? :)
Thank you in advance 

[image]

dNechita, 2019-02-28:
Hi,

Please don't use cmake. Use directly the Makefile that is available in the repository. Most likely that after you ran cmake, the Makefile has been overwritten so make sure you use the initial Makefile. The cmake file that is there is not for Linux (yet), sorry for this confusion.

Can you also check which version of GNU libc your system has?

SarahWicker, 2019-02-28:
Hi,
I went back to the unmodified git version (without cmake) but still get the same error: 
[image]

Also the Version of GNU libc is 4.6.3... Maybe this is to old? 
[image]

Additionnal Question, the libiio and libad9361-iio, were built with cmake (and it worked) but maybe should be build with make?

Best regards

dNechita, 2019-02-28:
Your GNU libc version is 2.15 which is now almost 7 year old (https://sourceware.org/glibc/wiki/Glibc%20Timeline).
Is it possible to use newer software?

Regarding your additional question: both libiio and libad9361-iio make use of cmake to generate Makefile(on Linux) which then is used to do the actual build. We hope to do the same for iio-oscilloscope.

rgetz, 2019-03-01:
@SarahWicker : Things are tested on the ADI image - it might be easier to use that SD Card.

https://wiki.analog.com/resources/tools-software/linux-software/zynq_images

If you need the very latest of everything, use ```adi_update_tools dev```

https://wiki.analog.com/resources/tools-software/linux-software/zynq_images#staying_up_to_date

SarahWicker, 2019-03-04:
Hi 
Sorry for taking so long to answer. 
@dNechita i tried to upgrade several time the linaro distro that i was using but always ended up into dependancy Errors.. 
i try to use more recent Version via the ubuntu-build-service from linaro, haven't been able to test yet (some Problem on the build part) but hope this could work.
@rgetz, I tried with the ADI Image and it worked fine. 
In the end read the Version of the GNU libc and it is 2.19 so imagine that the previous one was to old like you said and couldn't manage the build of the iio-oscilloscope
Thank you again for your help! :)

closed 2019-03-11 by dNechita