analogdevicesinc/iio-oscilloscope #143: Feature Request: Add experimental VNA capability to iio-oscilloscope

satfan52, 2019-03-08:
The PlutoSSR board (70mhz -6ghz) and the red pitaya SDR (0-60 mhz) share "similar" hardware architecture based on the Zyncq 7010. 

The red pitaya and the PlutoSDR howebet uses different "front end" and "back-ends" to the Zyncq10; An AD9363 transceiver for Pluto and two pairs pf DACs/ADCs for the red-pitaya; the Pluto also has much less memory than the red-pitaya.  However, despite those differences, the common Zynq 7010 platorm opens a potential avenue for sharing open-source code between the two environments. Among all open sources applications written for the red-pitaya, the VNA application one is probably the most interesting one to end-users wishing to experiment with the PlutoSDR for making RF measurements between 70 Mhz and 6ghz. 

As ii-oscilloscope is the main measurement application for the PlutoSDR, it would make sense to add experimental VNA functionality to it.  The existing VNA open-source code developed for the red-pitaya could be used as a basis/starting point or source of inspiration for that purpose.

tfcollins, 2019-03-08:
Similar functionality is being developed in our Scopy application, which is planned to have PlutoSDR support in the future.
See doc on Scopy: https://wiki.analog.com/university/tools/m2k/scopy

satfan52, 2019-03-08:
Thanks, I see indeed that a network analyser functionality is foreseen, and if the PlutoSDR is supported, it will indeed be a very useful application for users wanting to make experimental rf-measurements with the Pluto. However, from what I see of the pics in the docs, not smith chart capabilities are foreseen at the moment. If it is not foreseen, it would be nice to add it for a true VNA experience, as well of course as  some calibration process (Open, Short, Load) and the possibility to set a reference plane.