analogdevicesinc/iio-oscilloscope #148: Wrong PLL chip reference

JanVerlinden, 2019-03-29:
If a ZCU102 board with a ADRV9009-W/PCBZ is used. The application will look for a HMC7044. Which is not on the evaluation board. On this board a AD9528 is used.

mhennerich, 2019-03-29:
Besides an annoying error message, this should not affect things.
But you are right:
MCS should only be done displayed when more then one ADRV9009 is present.
And the error should be less verbose.

In general this MCS is only a temporary workaround until the JESD204 interface framework can handle it internally, without user space application support.

We fix the issue soon.

JanVerlinden, 2019-03-29:
I would prefer that this function would be available for the AD9528 as well.
I am using 2 platforms, which I need to sync together. Each platform has it own, ZCU102 board, ADRV9009-W/PCBZ board and its own iio-oscilloscope application running.
Moving it to the JESD204 frame work will not fix it, I think. Because the ADRV9009 will need it for RfLOPhaseSync.

commodo, 2019-04-02:
@JanVerlinden 
the change has been merged to master & 2018_R2

if no objections, let's close this
feel free to re-open if still an issue

thanks
Alex

closed 2019-04-02 by commodo