analogdevicesinc/iio-oscilloscope #153: Executable will not run on Windows 10

walshal, 2019-04-11:
I installed the version 0.10 windows executable on Windows 10. When i try to run iio-scope i get the following message
[image]

rgetz, 2019-04-17:
I don't get this error - I just tried it. 

Are you on Win32, or Win64?

walshal, 2019-04-17:
Win64. I noticed I only get this error when my laptop is plugged into dual external monitors. If I open iioscope on the laptop monitor it opens correctly. Seems to be linked to display settings.

mhennerich, 2019-04-18:
Can you try to delete following file?

C:\Users\YOUR-USER\AppData\Local\.osc_profile.ini

OSC saves the last position of the window and if that position doesn't exist (because it's currently outside the available desktop area) undesired behavior might exist. We have a look and fix this soon.

-Michael

mhennerich, 2019-04-18:
@adisuciu can someone have a look?

walshal, 2019-04-19:
I couldn't find that file in the location you stated. I did find it in c:\tmp but deleting it didn't help. Iioscope would not open when monitors plugged in.

adisuciu, 2019-04-22:
2e5ca7448129f4b0cf5c39a65b5b9b5dfde24674 should fix it

closed 2020-06-22 by mhennerich