analogdevicesinc/iio-oscilloscope #155: Make install problem when using cmake

nunojsa, 2019-04-12:
When using cmake, make install is falling. This is happening because `ldconfig` is not being executed after installation, leading to:
```
nsa@nsa-Precision-5530:~/work/iio-oscilloscope$ osc
osc: error while loading shared libraries: libosc.so.0: cannot open shared object file: No such file or directory 
```
So, I suggest we either make it happen during the install process or document it and say (hey, after make install you must run `sudo ldconfig`). Alternatively, we can use rpath/runpath...

nunojsa, 2021-08-06:
This can be closed...

closed 2021-08-06 by nunojsa

JPHutchins, 2023-01-24:
I did not see this in the documentation here: https://wiki.analog.com/resources/tools-software/linux-software/libiio#how_to_build_it

Thanks for checking!