analogdevicesinc/iio-oscilloscope #167: triggers in osc are broken

rgetz, 2019-07-25:
They no longer work - I didn't go back to find out who broke this or when.

mhennerich, 2019-07-25:
cannot reproduce - can you share your test settings?

rgetz, 2019-07-25:
Sure - it looks like it is related to the number of samples...

(Digital loopback) 128 samples works:

[image]

127 samples does not.
[image]

129 samples does not.
[image]

126 works
125 does not

looks like all odd sample numbers don't work (on Pluto).

PR #183 (2019-09-23): Trigger fixes
This fixes #167
closed 2019-09-23 by commodo