analogdevicesinc/iio-oscilloscope #168: AD9361 DDSes are broken

rgetz, 2019-07-25:
You should be able to control the AD936x DSSes independently  - you no longer can. I think when the HDL updates were made to the DDS - something got broken.

Setup: On ADALM-PLUTO, set up digital loopback:
[image]

Set the DDS to only be one signal on I, and nothing else:
[image]

You still see two signals:
[image]

This is a bug either in osc (the way it writes to the driver), in the driver (the way it writes to the HDL) or in the HDL itself.

rgetz, 2019-07-25:
When things are 90 degrees out of phase, it works great:
[image]
results in:
[image]

changing to the same phase:
[image]

doesn't change the output properly.
[image]

rgetz, 2019-07-25:
# iio_attr -C
IIO context with 7 attributes:
hw_model: Analog Devices PlutoSDR Rev.B (Z7010-AD9364)
hw_model_variant: 0
hw_serial: 104473222a87000618000600473ed57ae0
fw_version: v0.31
ad9361-phy,xo_correction: 40000000
ad9361-phy,model: ad9364
local,kernel: 4.14.0-42540-g387d584

osc is master

libiio is 

rgetz@brain:~/github/libiio/build$ iio_info 
Library version: 0.18 (git tag: c4833c5)
Compiled with backends: local xml ip usb serial

mhennerich, 2019-07-25:
Hmm - what version of OSC are you using?
I can not reproduce this issue at all, using OSC from master, latest libiio and pluto-fw.

Also the triggering works for me.

rgetz, 2019-07-25:
Doc is master from yesterday

mhennerich, 2019-07-25:
I'm also using latest master builds. Could it be that your libiio is dirty? Can you try a fresh checkout?

rgetz, 2019-07-25:
Looks like I accidentally set the Rx Rotation.

[image]

Yes - things are working as expected - sorry for the noise - I guess I shouldn't post bugs when it's late...

closed 2019-07-25 by rgetz