analogdevicesinc/iio-oscilloscope #178: Multiple windows partially broken with ini tests

tfcollins, 2019-08-02:
For FMComms5 ini tests two capture windows are launched during test. One is done by the ini file manually, the second appears when doing calibration. However, after the calibration window launches remaining capture windows cannot be resumed. This is only fixed by runnin a fru_connect which requires user interaction.

To repeat run standard FMComms5_test.ini, which should fail after: https://github.com/analogdevicesinc/iio-oscilloscope/blob/9b9a4410be713a3ce2d72b5bac07f225ca33aef7/profiles/FMComms5_test.ini#L240

dNechita, 2019-11-15:
I couldn't find the issue you've mentioned. What do you mean by fail after 'calibrate = 1'? Is there an error message? Or does something get stuck or crashes? Can you provide more details about this? What is expected to work but doesn't?

tfcollins, 2019-11-15:
From what I recall it gave a test failure.
To be more specific I was using the production script https://github.com/analogdevicesinc/linux_image_ADI-scripts/blob/master/test_fmcomms5.sh
running on the board itself.