analogdevicesinc/iio-oscilloscope #185: iio-oscilloscope crashes when using the spectrum analyzer over the network

satfan52, 2019-10-01:
Hello,

I have compiled iio-oscilloscope from the beeding edge master branch in my Ubuntu 18.10 environment and it works quite well when my pluto is connected to my PC via the USB port. I can use the spectrum analyzer functionality without any issues.

If connect my pluto via the network using an external Ethernet adapter I can use all the functionalities of iio-oscilloscope EXCEPT the spectrum analyzer functionality. When I launch the spectrum analyzer iio-oscilloscope immediately crashes afterward (see crash report below). This strange because everything else works fine in network mode with an external Ethernet adapter, including frequency domain analysis,  but when the spectrum analyzer functionality is started then iio-oscilloscope immediately crashes. 

IIO-Oscilloscope Crash Info

PID: 30209
Signal No: 11
Signal String: Segmentation fault
Error No: 11
Error String: Resource temporarily unavailable
Time Stamp: Tue Oct  1 23:49:17 2019

IIO-Oscilloscope Backtrace
osc(obtainBacktrace+0x88)[0x555e4bedac8e]
/lib/x86_64-linux-gnu/libc.so.6(+0x3ef20)[0x7fad10f1ef20]
/usr/lib/x86_64-linux-gnu/libiio.so.0(iio_channel_attr_write_raw+0x6)[0x7fad11537256]
/usr/lib/x86_64-linux-gnu/libiio.so.0(iio_channel_attr_write_longlong+0x56)[0x7fad11537d06]
/usr/local//lib/osc//spectrum_analyzer.so(+0x3f2a)[0x7facf6653f2a]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_closure_invoke+0x19d)[0x7fad1176010d]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x2305e)[0x7fad1177305e]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0xa75)[0x7fad1177b715]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f)[0x7fad1177c12f]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x8d6f5)[0x7fad11f156f5]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_closure_invoke+0x19d)[0x7fad1176010d]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x2312e)[0x7fad1177312e]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0xa75)[0x7fad1177b715]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f)[0x7fad1177c12f]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x8c679)[0x7fad11f14679]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x13238b)[0x7fad11fba38b]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_closure_invoke+0x19d)[0x7fad1176010d]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(+0x22de8)[0x7fad11772de8]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit_valist+0x40f)[0x7fad1177b0af]
/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0(g_signal_emit+0x8f)[0x7fad1177c12f]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(+0x2482bc)[0x7fad120d02bc]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_propagate_event+0xac)[0x7fad11fb862c]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_main_do_event+0x2fb)[0x7fad11fb8a2b]
/usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0(+0x5b02c)[0x7fad11c2b02c]
/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2e7)[0x7fad1251c417]
/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4c650)[0x7fad1251c650]
/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_loop_run+0xc2)[0x7fad1251c962]
/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0(gtk_main+0xb7)[0x7fad11fb7a37]
osc(main+0x22e)[0x555e4bedbe66]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7)[0x7fad10f01b97]
osc(_start+0x2a)[0x555e4bedaaea]

dNechita, 2019-10-02:
Hi,

I will look into this. Thanks for the crash report.

dNechita, 2019-10-02:
I wasn't able to reproduce this on a first try. IIO-Oscilloscope depends heavily on libiio. Can you share which version of libiio are you using? Also, please make sure you are using the latest firmware on Pluto.

If it's not much trouble, can you run osc with gdb? It provides a more detailed backtrace that the crash report. Just run `gdb ./osc` and then enter the `r` command. And post that backtrace.

satfan52, 2019-10-02:
Dear Dan,
libiio version is 0.18,  compiled from source code available on github
I use the latest version of the firmware (0.31) on the pluto
Will run gdb and post results here

satfan52, 2019-10-02:
I tried  "gdb ./osc" but gdb complains about the lack of debugging symbols. Should I compile iio-oscilloscope with specific parameters to be able to use gdb ?

This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./osc...(no debugging symbols found)...done.
(gdb)

dNechita, 2019-10-03:
To include the symbols, usually the -g option needs to be added to the gcc but this already done in the Makefile. So it should had worked. What could help is to edit the Makefile and replace the -O2 option that is given to CFLAGS and overwrite it with -ggdb, then clean and build again.

dNechita, 2019-10-03:
If by any chance you've been building with CMake, try using the existing Makefile instead.

satfan52, 2019-10-03:
Hello,

Here are the answers to the questions

1) I use the sequence "make clean", "cmake .", "make", "make install"

2) There is a Makefile but it does not contain any O2 and CFLAGS parameters, so it is useless. I don't really think I could us make instead of cmake and anyway I found a way of setting the right options in cmake (see point 3)

3) cmake generates a CMakeCache.txt file and in this file I found the -O2 option and replaced it with  -ggdb. I think it work (see point 6 with the gdb debuging info). 

4) I think the reason gdb did not work initially is because in the  CMakeCache.txt, the -g option is not set for every types of builds, only for specific ones (there is a very long list !).

5) Something very strange happens after I compile. When I launch ./osc from the directory where I compiled it (/opt/build/iio-oscilloscopeà, the spectrum analyzer tab does not appear....as if the plugin did not load... if I launch osc from the home directory with the command "/opt/build/iio-oscilloscope/osc" then the spectrum analyzer tab appears. I don't understand what happens

6) please find below the results of the "gdb  /opt/build/iio-oscilloscope/osc" command, you can see the crash after I launch the spectrumanalyzer

root@on7yi-ubuntu:~# gdb /opt/build/iio-oscilloscope/osc
GNU gdb (Ubuntu 8.1-0ubuntu3.1) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/build/iio-oscilloscope/osc...(no debugging symbols found)...done.
(gdb) r
Starting program: /opt/build/iio-oscilloscope/osc 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe4188700 (LWP 25090)]
[New Thread 0x7fffe25e2700 (LWP 25091)]
[New Thread 0x7fffe1de1700 (LWP 25092)]
[New Thread 0x7fffe15e0700 (LWP 25093)]
[Thread 0x7fffe15e0700 (LWP 25093) exited]
Found plugin: AD936X
Found plugin: AD6676
Found plugin: CN0508
Found plugin: FMCOMMS11
Found plugin: AD936X Advanced
Found plugin: CN0357
Found plugin: FMComms6
Found plugin: LIDAR
Found plugin: Debug
Found plugin: DMM
Found plugin: ADRV9009
Found plugin: Motor Control
Found plugin: FMComms5
Found plugin: Partial Reconfiguration
Found plugin: FMComms1
Found plugin: SCPI
Found plugin: AD9371 Advanced
Found plugin: AD5628-1
Found plugin: FMCADC3
Found plugin: AD7303
Found plugin: ad9739a
Found plugin: AD9371
Found plugin: Spectrum Analyzer
Found plugin: DAQ1/2/3
[New Thread 0x7fffe15e0700 (LWP 25094)]
[New Thread 0x7fffd3daa700 (LWP 25095)]
[New Thread 0x7fffd35a9700 (LWP 25096)]
[New Thread 0x7fffd2da8700 (LWP 25097)]
[New Thread 0x7fffd25a7700 (LWP 25098)]
[Thread 0x7fffd2da8700 (LWP 25097) exited]
[Thread 0x7fffd25a7700 (LWP 25098) exited]
[Thread 0x7fffd35a9700 (LWP 25096) exited]
[New Thread 0x7fffd1da6700 (LWP 25099)]
Updating widgets...
[Thread 0x7fffd3daa700 (LWP 25095) exited]
Updating FIR filter...
[Thread 0x7fffe15e0700 (LWP 25094) exited]
Loaded plugin: AD936X
Loaded plugin: AD936X Advanced
Loaded plugin: Debug
Loaded plugin: DMM
Loaded plugin: Spectrum Analyzer
[New Thread 0x7fffd25a7700 (LWP 25100)]
[New Thread 0x7fffd2da8700 (LWP 25101)]
[Thread 0x7fffd2da8700 (LWP 25101) exited]
[New Thread 0x7fffd2da8700 (LWP 25102)]
[Thread 0x7fffd2da8700 (LWP 25102) exited]
[Thread 0x7fffe1de1700 (LWP 25092) exited]
[Thread 0x7fffe25e2700 (LWP 25091) exited]
[Thread 0x7fffd25a7700 (LWP 25100) exited]
[Thread 0x7fffd1da6700 (LWP 25099) exited]

Thread 1 "osc" received signal SIGSEGV, Segmentation fault.
0x00007ffff6b39256 in iio_channel_attr_write_raw (chn=0x7fffc000c260, 
    attr=0x7fffe001b626 "frequency", src=0x7fffffffcfd0, len=11)
    at /usr/src/libiio/channel.c:364
364		if (chn->dev->ctx->ops->write_channel_attr)
(gdb)

satfan52, 2019-10-05:
Is this helpful ?

satfan52, 2019-10-05:
Out of curiosity I did the test in a windows environment, under Windows 7 in a virtual box environment. The Ethernet interfaces is shared in a bridged way (no NAT) so the Plutosdr and the Windows 7 virtual PC are on the same subnet. Surprisingly the spectrum analyzer tab does NOT show up... Reading the docs of the spectrum analyzer plugin, it seems that if the right type of hardware board is not detected the spectrum analyzer plugin does not load. Maybe the problem is related to this detection

mhennerich, 2019-10-07:
Please see here: https://github.com/analogdevicesinc/iio-oscilloscope/commit/68cb9095987897760d1ad44b4b0abc463cb0c818

dNechita, 2019-10-07:
The crash report that you provided with gdb is helpful, thanks! It would be of most help if you could send the backtrace as well. When the segmentation fault is caught  by gdb, just type `bt full` to get the full backtrace.
Thanks!

satfan52, 2019-10-07:
Hello ;-)
Here is the full trace (with but full) as per your request.
Thanks !

root@on7yi-ubuntu:~# gdb osc
GNU gdb (Ubuntu 8.1-0ubuntu3.1) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from osc...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/local/bin/osc 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe4188700 (LWP 20988)]
[New Thread 0x7fffe25e2700 (LWP 20989)]
[New Thread 0x7fffe1de1700 (LWP 20990)]
[New Thread 0x7fffe15e0700 (LWP 20991)]
[Thread 0x7fffe15e0700 (LWP 20991) exited]
Found plugin: AD936X
Found plugin: AD6676
Found plugin: CN0508
Found plugin: FMCOMMS11
Found plugin: AD936X Advanced
Found plugin: CN0357
Found plugin: FMComms6
Found plugin: LIDAR
Found plugin: Debug
Found plugin: DMM
Found plugin: ADRV9009
Found plugin: Motor Control
Found plugin: FMComms5
Found plugin: Partial Reconfiguration
Found plugin: FMComms1
Found plugin: SCPI
Found plugin: AD9371 Advanced
Found plugin: AD5628-1
Found plugin: FMCADC3
Found plugin: AD7303
Found plugin: ad9739a
Found plugin: AD9371
Found plugin: Spectrum Analyzer
Found plugin: DAQ1/2/3
[New Thread 0x7fffe15e0700 (LWP 20992)]
[New Thread 0x7fffd3fff700 (LWP 20993)]
[New Thread 0x7fffd37fe700 (LWP 20994)]
[New Thread 0x7fffd2ffd700 (LWP 20995)]
[New Thread 0x7fffd27fc700 (LWP 20996)]
[Thread 0x7fffd2ffd700 (LWP 20995) exited]
[Thread 0x7fffd27fc700 (LWP 20996) exited]
[Thread 0x7fffd37fe700 (LWP 20994) exited]
[New Thread 0x7fffd1ffb700 (LWP 20997)]
[Thread 0x7fffd3fff700 (LWP 20993) exited]
Updating widgets...
Updating FIR filter...
[Thread 0x7fffe15e0700 (LWP 20992) exited]
Loaded plugin: AD936X
Loaded plugin: AD936X Advanced
Loaded plugin: Debug
Loaded plugin: DMM
[... 30 lines cut ...]
#5  0x00007ffff6d5f10d in g_closure_invoke () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#6  0x00007ffff6d7205e in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#7  0x00007ffff6d7a715 in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#8  0x00007ffff6d7b12f in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#9  0x00007ffff75096f5 in  () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#10 0x00007ffff6d5f10d in g_closure_invoke () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#11 0x00007ffff6d7212e in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#12 0x00007ffff6d7a715 in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#13 0x00007ffff6d7b12f in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#14 0x00007ffff7508679 in  () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#15 0x00007ffff75ae38b in  () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#16 0x00007ffff6d5f10d in g_closure_invoke () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#17 0x00007ffff6d71de8 in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#18 0x00007ffff6d7a0af in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#19 0x00007ffff6d7b12f in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#20 0x00007ffff76c42bc in  () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#21 0x00007ffff75ac62c in gtk_propagate_event () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#22 0x00007ffff75aca2b in gtk_main_do_event () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#23 0x00007ffff722202c in  () at /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#24 0x00007ffff7b0a417 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#25 0x00007ffff7b0a650 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#26 0x00007ffff7b0a962 in g_main_loop_run () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#27 0x00007ffff75aba37 in gtk_main () at /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#28 0x0000555555557e66 in main ()
(gdb)

dNechita, 2019-10-08:
Hi,

There are 2 ways to build iio-osc. One is to use the existing Makefile which is part of the repo and is the official method. The second is using CMake.
it seems that using CMake, the Spectrum Analyzer plugin is omitted from being built. This means that when you built osc and all plugins, it didn't build spectrum_analyzer.c. Since your instance of iio-osc did load the spectrum analyzer, it means that you have it build on your system. It might be built with an older version of libiio, not sure. I've made some changes so that the CMake approach includes the plugin.
Can you try building iio-osc again, from scratch, using this branch (https://github.com/analogdevicesinc/iio-oscilloscope/tree/cmake-include-all-plugins)?

Thanks!

satfan52, 2019-10-08:
Hello,

Thanks for the explanation, it is getting clearer.

1) I found THREE other iio-oscilloscope binaries under /usr/local/bin, they are git version g7fa5158, g6e76199 and g9159e07. I don't remember how they got there as I compile everything under /opt/build/iio-oscilloscope nowadays. So in total I have 4 different iio-oscilloscope binaries installed on my system :-(

2) Under /usr/local/lib I found the spectrumanalyser.so plugin, I am sure it belongs to an older version because iwhen I compile the current version from the /opt/build/iio-oscilloscope directory there is indeed NO spectrumanalyser.so plugin available in the directory /opt/build/iio-oscilloscope/plugins

3) when run ./osc within /opt/build/iio-oscilloscope directory,  the specttrumanalyzer.so plugin is NOT found, but if I run the full command /opt/build/iio-oscilloscope/.osc from any OTHER directory then the plugin in the spectrumanalyser.so plugin is found from /usr/local/lib. I am shure it is the only copy available in name because if I rename it then no plugin is found.

4)  I cloned the branch you indicated me and then did the sequence cmake. , make , make install but nothing, there is no change, the behaviour is exactly the same than the one of the master branch. The file spectrumanalyzer.so is NOT created in the  /opt/build/iio-oscilloscope/plugins directory either

5) Unless I am mistaken, there is no Deb package available that I can download and install with the Ubuntu packet manager so I am not shure I follow you there. If there would be one, that method would be fine with me as long as the spectrumanalyszer plugin is created. I am of course willing to use the "Make" method if cmake is the wrong one, in this case please point me to relevant instructions.

Many thanks for your help  
Regards
Peter

satfan52, 2019-10-08:
Ok, I have used this method:

https://github.com/analogdevicesinc/linux_image_ADI-scripts/blob/master/adi_update_tools.sh

the plugin spectrumanalyzer.so got created successfully in /usr/local/lib/osc/plugins and I can use it successfully over the network ;-) ;-) ;-) So I guess I am a happy man even  if I don't undetstand what was the problem at the first place with the cmake approach.

Many Thanks

closed 2019-10-08 by satfan52

mhennerich, 2019-10-09:
Looks like the Makefile and CMake option will install files into different paths.
We need to switch completely to the CMake approach, but need to make sure all other shared libraries get removed.

dNechita, 2019-10-09:
Hi Peter,

Our cmake is in an experimental state right now but I will get better.

The weird behavior of the spectrum_analyzer plugin when using the ethernet adapter can be explained by the fact that having the plugin built with an old enough version of libiio can cause undefined behavior which is the realm of anything can happen :)

I'm glad things are now working for you.