analogdevicesinc/iio-oscilloscope #194: DDS Mode: set scale(dBFS) is not working under mac os x on an Adalm Pluto board and no tone is output by the board

sacatak, 2019-10-24:
When I set on my mac machine the parameter scale(dBFS) in DDS Mode for "one cw tone" or "two cw tones", I can't see any tone outputs from the board.
When I do the same configuration from my ubuntu machine, it works. Both machines have the same version: master-g8825ea7 and libiio-version:0.18-c0012d0
I have found a workaround under mac by using debug>cf-ad9361-dds-core-lpc and writing directly the iio device attributes like output atlvoltage0(TX1-I_F1) scale to outputs the tone.
I have the feeling that setting the scale of the tones by the DDS mode window do nothing.
In addition when I push "reload settings", I see that these scale values are always refresh with -inf value since I put a value directly in the iio device attributes

tfcollins, 2019-10-24:
I can't repeat this issue on Mojave with master or the latest release.
The version of macOS are you on? Have you tried the brew builds?

-Travis

sacatak, 2019-10-25:
Thanks Travis for your return. My mojave version is 10.14.6.
I'm not too familiar with brew.
I've installled the software with the following command:  brew install --HEAD tfcollins/homebrew-formulae/i-i-o-oscilloscope. As it seems that it was probably a software issue, I've edit the brew with "brew edit tfcollins/formulae/i-i-o-oscilloscope" to test the branch "2019_R1" by adding: ,:branch => "2019_R1" at the end of line head "https://github.com/analogdevicesinc/iio-oscilloscope.git".
And I have the same behavior with the 2019_R1 version.
I don't understand what you mean by trying the brew builds.
Concerning the libad9361, I've installed the latest: master_latest_libad9361-iio-darwin-10.14.4.pkg 
Do you have other ideas to investigate.

Regards,
Nicolas

tfcollins, 2019-10-25:
Try the Mac instructions here: https://wiki.analog.com/sdrseminars

-Travis

sacatak, 2019-10-25:
Dear Travis,

I've followed the instructions from https://wiki.analog.com/sdrseminars on an other mac computer and I've still have the problem.
Regards,
Nicolas

sacatak, 2019-10-25:
[Oct-25-2019 22-33-44.mp4.zip](https://github.com/analogdevicesinc/iio-oscilloscope/files/3773721/Oct-25-2019.22-33-44.mp4.zip)

Dear Travis,
In this video, you'll see that at the begin of the video I try to change the scale of the tone and nothing happens. When I push reload settings the scale pass to -inf. After that I put directly the value on the register and the spectrum shows the tone. When I come back on the scale, it has been updated
Regards,
Nicolas

mhennerich, 2019-10-25:
What LOCALE settings are you using?

tfcollins, 2019-10-25:
@mhennerich is probably right here. We've seen similar things with non-English installs

-Travis

sacatak, 2019-10-26:
> What LOCALE settings are you using?

Thanks @mhennerich for your your answer.
My locale settings are french

sacatak, 2019-10-29:
Dear @tfcollins ,
Do you think is it possible to correct this behaviour?
Regards,
Nicolas

tfcollins, 2019-10-29:
This can be difficult for us to support since no current staff use that locale. We do accept full requests if you find a solution:). I will defer to @mhennerich though.

-Travis

mhennerich, 2019-10-30:
The problem is really that the kernel has a fixed API. For IIO the scales use a period to indicate the decimal place. It is a very common problem, but libiio is intended to work around it by providing 
methods such as iio_device_attr_read_double() to mitigate this problem.
The issue persists if for exmple iio_device_attr_read() is used and then the string maybe containing a number is parsed using scanf() and friends. There seems to be such places dac_data_manager and and it's underlying iio_widget library.

Maybe @dNechita  can have a look and see if there is a simple fix.

-Michael

dNechita, 2019-10-31:
It's very likely that parts of iio-osc are interpreting float strings in a way which ignores LOCALE. I'll look into it.
But in this case I can track the problem all the way to libiio, were iio_device_attr_write_double() doesn't work as expected.
The internal implementation of iio_device_attr_write_double() will use iio_snprintf() which is actually snprintf. This happens because on macOS the following never gets defined:
https://github.com/analogdevicesinc/libiio/blob/v0.18/utilities.c#L33
Therefore, the snprintf will format a string using a comma instead of dot which is specific to a number of locales. E.g "0,44". When this string if passed to iio_channel_attr_write(), it fails with Invalid Argument as error.

dNechita, 2019-11-05:
A few more details about this:
Simple programs based on libiio (e.g iio_info, iio_readdev, etc.) work well on mac OSX because they get a default locale (which is called locale C) no matter what locale is set on the system.
What happens for iio-oscilloscope is that it is also based on GTK2 which is a UI library that needs to be aware of locale. In iio-oscilloscope, when gtk_init() gets called, internally, the locale is being set to whatever the system locale is. From this point on everything is being affected by locale, including libiio.

closed 2019-11-05 by dNechita